raspiblitz/raspiblitz #4842: Update to 1.11.3 issues / RPi4 / c-lightning

hermanlim, 2024-11-10:
**Describe the bug**
After upgrading from 1.11.1, lightningd seems to be unable to connect to the bitcoin RPC causing the "stuck at 99.96%". Prior to upgrade, system was working well with occasional issues with the gossip file, easily fixed by deleting it and having the system create a new one.

Any pointers to help resolve would be appreciated :) 

**To Reproduce**

**Debug file**
----I believe this is the relevant bit, the entire file is included below.

2024-11-10T13:56:48.861Z **BROKEN** lightningd: The Bitcoin backend died.
2024-11-10T13:56:49.474Z **BROKEN** lightningd: FATAL SIGNAL 6 (version v24.08.1-modded)
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: common/daemon.c:38 (send_backtrace) 0x557f9fdbdf
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: common/daemon.c:75 (crashdump) 0x557f9fdc33
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: (null):0 ((null)) 0x7f8cb767af
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ./nptl/pthread_kill.c:44 (__pthread_kill_implementation) 0x7f8c7409f0
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ../sysdeps/posix/raise.c:26 (__GI_raise) 0x7f8c6fa72b
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ./stdlib/abort.c:79 (__GI_abort) 0x7f8c6e747b
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: lightningd/log.c:1038 (fatal_vfmt) 0x557f9a34d3
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: lightningd/log.c:1048 (fatal) 0x557f9a3563
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: lightningd/bitcoind.c:36 (bitcoin_destructor) 0x557f97326b
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:246 (notify) 0x557faf1ab7
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:437 (del_tree) 0x557faf1ba3
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:532 (tal_free) 0x557faf21b7
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: lightningd/plugin.c:459 (plugin_kill) 0x557f9c9227
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: lightningd/plugin.c:890 (plugin_conn_finish) 0x557f9c954b
2024-11-10T13:56:49.474Z **BROKEN** lightningd: backtrace: ccan/ccan/io/poll.c:246 (destroy_conn) 0x557fae64ab
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/io/poll.c:252 (destroy_conn_close_fd) 0x557fae64d7
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:246 (notify) 0x557faf1ab7
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:437 (del_tree) 0x557faf1ba3
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/tal/tal.c:532 (tal_free) 0x557faf21b7
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/io/io.c:496 (io_close) 0x557fae4bdf
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ccan/ccan/io/poll.c:459 (io_loop) 0x557fae6c6f
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: lightningd/io_loop_with_timers.c:22 (io_loop_with_timers) 0x557f99aa77
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: lightningd/lightningd.c:1470 (main) 0x557f9a0937
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ../sysdeps/nptl/libc_start_call_main.h:58 (__libc_start_call_main) 0x7f8c6e773f
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: ../csu/libc-start.c:360 (__libc_start_main_impl) 0x7f8c6e7817
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: (null):0 ((null)) 0x557f971e2f
2024-11-10T13:56:49.482Z **BROKEN** lightningd: backtrace: (null):0 ((null)) 0xffffffffffffffff

*****Entire Log*******
Generating debug logs. Be patient, this should take maximum 2 minutes ..
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.11.3
commit-release: 59157c28
commit-active: 59157c28
chainnetwork: bitcoin / main
 21:28:15 up  7:42,  3 users,  load average: 4.33, 3.81, 3.63

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
[... 483 lines cut ...]
ln_default_wallet_onchain_balance=****
ln_default_channels_inactive="2"
ln_default_wallet_channels_pending="3034629"
ln_default_sync_progress="99.96"
ln_default_channels_active="9"
btc_default_blocks_data_kb="682574436"
btc_default_ready="1"
btc_default_blocks_verified="869768"
btc_default_sync_initial_done="1"
btc_default_activated="1"
btc_default_running="1"
btc_default_mempool_transactions="51115"
btc_default_error_short=""
btc_default_blocks_headers="869768"
btc_default_error_full=""
btc_default_peers="24"
btc_default_online="1"
btc_default_blocks_behind="0"
btc_default_sync_initialblockdownload="0"
btc_default_synced="1"
btc_default_version="v26.0.0"
btc_default_port="8333"
btc_default_sync_progress="0.9999989296447813"
btc_default_sync_percentage="100.00"
*** POSSIBLE ERROR REPORTS ***

codastrian, 2024-11-11:
I can confirm that I experience the same issues. Started with RaspiBlitz "1.11.2" for me. 
Tried to manually upgrade to CLN "v24.08.2" but didn't fix the issue.
Now running 1.11.3 with CLN 24.08.1 again, but still the same issue.

hermanlim, 2024-11-11:
@codastrian 
I entered the menu manually and set up password for the c-lightning wallet. Interestingly, this act solved the problem for me!

rootzoll, 2024-11-12:
@openoms can you take a look?

codastrian, 2024-11-12:
I re-flashed my Pi and set it up from scratch, only restoring the CLN from the recovery tar that you can create. So far it seems to work fine. I will update the item if I experience this again.

fossil, 2024-12-11:
I have the same issue after I upgraded to 1.11.3 from 1.10. 
Sync 99.9%.
..."STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""...

hermanlim, 2024-12-11:
Updating to latest version of Bitcoin + CL seemed to fix the issue for me.I still get an error where CL refuses to start properly sometimes but that is fixed by deleting gossip file and restarting. Please excuse the brevity and any typos as this message has been composed on a mobile device. iMessage public key: APKTIDPgMjMi-v57NPO7t0epNrMFvRLjrvj6qMbKi_2SACiQSWHw

rootzoll, 2024-12-12:
@openoms would a CLN upadate (with old Bitcoin Core base) for v1.11.4 help here?

rootzoll, 2024-12-18:
RaspiBlitz v1.11.4 will come with Core Lightning v24.11 - after upcomming Release please test if fixed. Will close issue until then.

closed 2024-12-18 by rootzoll