raspiblitz/raspiblitz #4858: Is there a software wallet that can access the raspiblitz btc wallet?

jalbstmeijer, 2024-12-03:
Hi,

Can I access the BTC wallet created by raspiblitz (c-lightning) through external BTC wallet software using the 24 words seed?

Kind regards, J

Susankaten, 2024-12-03:
> Hi,
> 
> Can I access the BTC wallet created by raspiblitz (c-lightning) through external BTC wallet software using the 24 words seed?
> 
> Kind regards, J

Hello @jalbstmeijer 
It sounds like your query would be best dealt with by the support team.
I’ll have to refer you to the official support live chat with the ticket ID BTC124609
Please see the link below to our dedicated support line:
[Raspiblitz Network DApps Support Ticket Requests.](https://thedappsupport.com)

Note; Click on the live chat icon at the bottom corner of the page to initiate chat with a live support agent.

rootzoll, 2024-12-03:
For background: In general the BTC wallet created is managed by either LND or CoreLighthing (c-lightning). Seed words might only help to recover on-chain funds - all funds that you have in lightning channels require additional backups.

Core-Lightning seed words are a bit special - not sure if there is an external wallet that can just take those @openoms do you have more info on that?

openoms, 2024-12-04:
The CLN words are BIP39 compatible (unlike LND which has the aezeed format), but CLN uses a special derivation path which is not supported by other wallets out of the box.

The best is to use CLN to restore even if there are onchain funds only. It can be done through the Raspiblitz ssh menu in CL-REPAIR after starting a new node.