raspiblitz/raspiblitz #4861: lnbits doesnt still due to postgress failure

warioishere, 2024-12-07:
raspiblitz 1.11.3

Trying to install lnbits fails with:

Assertion failed on job for postgresql@13-main.service.

making it not to start. 

Locales where not set properly.

Solution:

```
export LANG=en_US.UTF-8
export LC_ALL=en_US.UTF-8
export LANGUAGE=en_US.UTF-8
```

`sudo nano /etc/default/locale`

```
LANG="en_US.UTF-8"
LC_ALL="en_US.UTF-8"
LANGUAGE="en_US.UTF-8"
```

sudo locale-gen

sudo dpkg-reconfigure locales

be sure en_US.UTF-8 is selected

load en_US.UTF-8 in next step

start lnbits installation again, if it fails again. Do a

`sudo pg_createcluster 15 main`

and install again

rootzoll, 2024-12-12:
In version RaspiBlitz v1.11.3 there were still quotes missing on `LANG` value and `LANGUAGE` & `LC_ALL` are empty. 
```
admin@192.168.178.143:~ ₿ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

For version v1.12.0 its to recheck how the locale is finally set after the system is setup. There were some locale fixes alraedy in the build script but thats to re rechecked for v1.12.0

rootzoll, 2024-12-13:
We plan a smaller RaspiBlitz v1.11.4 release soon - basically just Alby-Hub & Bugfixes. 
So I added to the release script a part to again force the locales - this should than be part of v1.11.4.

But for v1.12.0 I crteated a new issue #4869 to move this forced locales over to the sd build card.

closed 2024-12-24 by rootzoll