raspiblitz/raspiblitz #4863: mempool install doesn't work using the minimal v1.11.3 lean image

openoms, 2024-12-10:
**Describe the bug**
mempool install doesn't work using the minimal v1.11.3 lean image
No problems when using the fat image.

**To Reproduce**
```
config.scripts/bonus.mempool.sh off
config.scripts/bonus.mempool.sh uninstall
config.scripts/bonus.mempool.sh on
```

**Debug file**
TBD

TrevorPhilipsBR, 2024-12-10:
I tried to reinstall mempool on 1.11.3 min without success. Log attached.

Reinstalled fat version (Menu, Update, Release , Start Update), but the result was the same.

Then I reinstalled fat version, using RECOVER option, and could install Mempool (Menu, Services, BTC Mempool Space)
[log install mempool.txt](https://github.com/user-attachments/files/18081394/log.install.mempool.txt)

RP Pi5
SSD Corsair 2TB
GeeekPi Raspberry Pi 5 N04 M.2 2280 PCIe to NVMe Top Expansion board Compatible with Raspberry Pi5

rootzoll, 2024-12-12:
@TrevorPhilipsBR from your logs this seems to be a network error - some downloads failed
@openoms I was not able to reproduce - mempool installs on my test machine

But I also saw multiple temp problems with downloads from github in the past - thats why a fatimage gets more important to reduce dependency on stable downloads from github.

Without further debug logs showing other causes I am closing this issue.

closed 2024-12-12 by rootzoll