raspiblitz/raspiblitz #4872: Proxmox VM Upgrade - Reboot Loop when HDD attached

curemici, 2024-12-14:
**Describe the bug**
When starting the the raspiblitz VM on Proxmox with the attached HDD with the blockchain data, the VM reboots after some seconds.

**To Reproduce**
1. REPAIR - MIGRATION, saved the migration file, stopped the raspiblitz-VM
Seen that it would better to use UPDATE, so 
2. Restart the raspiblitz VM - UPDATE - RELEASE - UPDATE NOW  - saved backup file - stopped the raspiblitz-VM with the old version
3. Prepared VM based on Debian 12.8 and build sdcard - first with branch v.11.1 and HDD (with blockchain and old data) attached
4. Restart raspiblitz VM - after about 10 seconds it reboots. When I manage to login I see the rebooting dialog counter. When I press cancel I get the message with the ssh login, but it continues with shutdown and reboot.

5. Prepared VM based on Debian 12.8 and build sdcard - now with branch dev and NO HDD attached. I could login but it asks for the HDD. As soon as I attach the HDD it also falls into the reboot loop.

(I've restarted my old raspi VM version again, so I can change on the HDD of do different preparing steps, if necessary)

rootzoll, 2024-12-17:
Hmm .. so there are some "reboots" in the boostrap script .. we would need to pin down which it is.

For this can you do the following:

- Boot up, connect/patchthru HDD and let reboot.
- Disconnect HDD on bootup
- Login per SSH, CTRL+C to terminal and `cat /home/admin/raspiblitz.last.log`
- Paste the last lines of that log file here for debugging

adevoss, 2025-02-04:
@Sep21-2009 Do you use a physical HDD? I never had this issue with a VM on Proxmox using two virtual disks.

Maybe this difference is relevant to solve this issue.