raspiblitz/raspiblitz #4874: First boot (pre installation) on Pi 4 doesn't proceed

21mln, 2024-12-15:
After writing the Raspiblitz image to the sd card and booting it from the Raspberry Pi 4 connected with hdmi, an extra 1TB external ssd and an ethernet cable, I get the same screen as #4559. Sometimes I get extra info where it says "THE FILE HAS BEEN MODIFIED".

At some point it did go through, where I ended up in the installation screen (blue background) where I saw "Starting Raspiblitz, please wait..." and from there it stayed in a loop and didn't continue. After this I haven't been able to enter this screen again.

I did found a workaround. When I get the same screen as #4559, I entered ALT+F2, logged in with pi/raspberry and the installation screen showed up (blue background), but it stayed in an andless loop stating "Starting Raspiblitz, please wait...", unfortunately.

Has somebody encountered this or perhaps know what the issue could be?

21mln, 2024-12-17:
> Hello @21mln
> 
> I got the same problem but only with the raspiblitz-fat-v1.11.3-2024-11-03.img.gz image.
> 
> The raspiblitz-min-v1.11.3-2024-11-03.img.gz boot is perfect.
> 
> I have not yet tried your work around.
> 
> Regards: peter

Thank you for pointing that out. I forgot to mention that I was using the fat version indeed. I will try the min version.

remicarreiro, 2024-12-26:
I have been experiencing this issue endlessly with version 1.11.4 and several previous versions that I've tested. This is my first time trying to set up a node so I'm still learning stuff. I am running a raspberry pi 5 with a pimoroni nvme base and 2TB Crucial P3 Plus drive. 

I kept getting stuck at the fsck screen but after seeing the ALT +F2 workaround, I was able to get past that screen and finally set things up after two days of trying various raspiblitz versions. I am currently syncing the blockchain, and not looking forward to having to deal with setting this thing up in its final place once it's all done. I just hope that I can SSH into it successfully when i reboot it.

Are there any plans to change the requirement for fsck at boot? Can this be omitted or is that a dumb move?

21mln, 2024-12-27:
I was also able to install the 1.11.4, but only after I did the keyboard shortcut ALT+F2 (or only F2) and then installation started. It's now syncing the blockchain.

closed 2024-12-30 by 21mln