raspiblitz/raspiblitz #4876: LNbits ADMIN-Switch

rootzoll, 2024-12-16:
`LNBITS_ADMIN_UI=true` is needed for initial setup, but by user (or automatically) it should be able to deactivate when done. 

Also when recovering LNbits the old `LNBITS_ADMIN_UI` value should be respected and not overwritten.

rootzoll, 2024-12-16:
OK release v1.11.4 will offer to switch AdminUI on/off from the RaspiBlitz SSH menus LNBITS > OPTIONS

 
[image]

[image]

warioishere, 2024-12-17:
I would add another button which sets the follwing line:

# Disable account creation for new users
`LNBITS_ALLOW_NEW_ACCOUNTS=
`
possible options `true` or `false`

for additional security purposes.

restart of lnbits is required for both actions to take effect.

greetings

warioishere, 2024-12-17:
> @warioishere First I was adding a "Allow new Accounts" option in SSH menus, but then realized that this option is available in the LNbits Admin section itself - so it does not make sense to replicate it.
> 
> <img alt="Bildschirmfoto 2024-12-17 um 13 37 50" width="1415" src="https://private-user-images.githubusercontent.com/2472225/396505804-c3af4370-b5a1-4447-8fa8-90d2252a22c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0NTQ2MzgsIm5iZiI6MTczNDQ1NDMzOCwicGF0aCI6Ii8yNDcyMjI1LzM5NjUwNTgwNC1jM2FmNDM3MC1iNWExLTQ0NDctOGZhOC05MGQyMjUyYTIyYzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTdUMTY1MjE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTI3MTM3NGM4ZmQ2NTI1Y2VlOTEzNzViMTljOTI3NTg5OWM1NzBkNzE2MGUxYzE3YzFmOTQwYmI1MTVhNmEyZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.weJhhbQC2rIm05ZpFNM5PnoMluksJ_i8LSAgaKBenXc">
> The SuperUser is created and set by LNbits by the `.superuser` file. I dont want to rush canhes here for zthe v1.11.4 .. but maybe the LNbits team @dni can recommend a better setup for v1.12.0 regarding if I should transfer the ID from `.superuser` to `.env` (using restart script) once its available.

yes you can enable user registration only when enabling the admin-ui, if you disable it, the server section is gone completly. 
I prefer to have it off, and then you can still disable and enable user registration without the need to activate the admin ui first before you can enable or disable user registration. 

about where the best place for the su ID, dni can give recommendation. I wasnt aware of the `.super_user` file sorry

rootzoll, 2024-12-17:
> yes you can enable user registration only when enabling the admin-ui, if you disable it, the server section is gone completly. I prefer to have it off, and then you can still disable and enable user registration with LNBITS_ALLOW_NEW_ACCOUNTS= without the need to activate the admin ui again before you can enable or disable user registration.

I would not like to replicate functionality that is alfready available in the LNbits UI - especially on when on first test the settings in the LNbits UI & .env did not sync ... that can all get messy quick with inconsistent state.

So I think its OK if you want to change the "Allow new User Registration" setting to enable WebAdmin once, make the change and the deactivate.

dni, 2024-12-19:
> Its not a replication of settings. The setting in the .env is ignored if Admin UI is enabled. The setting is activated if admin ui is disabled. See description if it:
> 
> ```
> # Enable Admin GUI, available for the first user in LNBITS_ADMIN_USERS if available.
> # Warning: Enabling this will make LNbits ignore most configurations in file. Only the
> # configurations defined in `ReadOnlySettings` will still be read from the environment variables.
> # The rest of the settings will be stored in your database and you will be able to change them
> # only  through the Admin UI.
> # Disable this to make LNbits use this config file again.
> ```
> 
> I always do it like that and it causes no inconstistant states at all, but of course, you are the boss and you take the decicion :)

you are right.

also now on the first installation when there is no superuser we now have this `first_install` page where you can set your superuser username and password.

rootzoll, 2024-12-19:
OK thats why it was out of sync while AdminUI was on - this is a bit confusing for users.

So here is my take away:
So only if the AdminUI is OFF it makes sense to give the option in the RaspiBlitz SSH LNbits menu to switch "Allow new User Registration" on or off, correct?

warioishere, 2024-12-19:
> OK thats why it was out of sync while AdminUI was on - this is a bit confusing for users.
> 
> So here is my take away: So only if the AdminUI is OFF it makes sense to give the option in the RaspiBlitz SSH LNbits menu to switch "Allow new User Registration" on or off, correct?

correct

rootzoll, 2024-12-19:
OK is implemented ans will be part of v1.11.4 - colsing issue for release

closed 2024-12-19 by rootzoll