raspiblitz/raspiblitz #4880: Add close all Channels to Lightning Backend Menu on ssh menu

warioishere, 2024-12-18:
As I serve in Raspiblitz German channel als support, I always overcome questions how to close channels all at once to get back funds to the internal wallet.

I would suggest adding another entry to the LND or CLN Entry in the menu with the possibilty to close all channels at once so people can easily use the Cashout function afterwards. 

on LND this would be lncli closeallchannels

```
OPTIONS:
   --inactive_only          close inactive channels only
   --force                  ask for confirmation once before attempting to close existing channels
   --conf_target value      (optional) the number of blocks that the closing transactions *should* confirm in, will be used for fee estimation (default: 0)
   --sat_per_vbyte value    (optional) a manual fee expressed in sat/vbyte that should be used when crafting the closing transactions (default: 0)
   -s, --skip_confirmation  Skip the confirmation prompt and close all channels immediately
```
   
 Fees can be provided after choosing the option on the ssh menu too.
 
 On cln, i dont know but I am sure can be done too, never used cln
 
 greetings

rootzoll, 2024-12-18:
I havent tested quite some time ... but is the CLOSEALL option not working anymore?

[image]

FYI the option is just displayed if you have open channels on LND. Maybe its an idea to remove the conditional displaying?

warioishere, 2024-12-18:
I have to ask the person again if this was really not displayed. Will get back to you

warioishere, 2024-12-18:
can be closed because of, option is there, some useres are maybe just blind, and hakuna ist working on a optimzed version which tries to close channels cooperatively

closed 2024-12-18 by warioishere