raspiblitz/raspiblitz #4884: SSH-Option "Close All Channels" - Try cooperative close first

rootzoll, 2024-12-18:
The "Close All Channels" option in den SSH menus calls the script: 
https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/BBcloseAllChannels.sh

But it just does a "force close" .. to try/offer a "Cooperative Close" first would be more economic aof course.

rootzoll, 2024-12-18:
@TrezorHannes  Tested the merged PR .. looks like if I choose "Cooperative Close" it will still ask on every channel point if I want to close Unilaterally (forced closed) - see:
 
[image]

So depending on state of channels (active, etc) it might ask for different things.

TrezorHannes, 2024-12-18:
Oh no, just figured `--force` has a different purpose in `closechannel` vs `closeallchannels`.

I will need to get back to the drawing board. There are two options to consider:
```
lncli closeallchannels
   --force                  ask for confirmation once before attempting to close existing channels
   -s, --skip_confirmation  Skip the confirmation prompt and close all channels immediately
```
So `force` is not even force-closing, it'll just ask for confirmation once, and then closing all: active as coop, inactive as force-close.
`skip_confirmation` does what it says, no confirmation needed at all and closes all immediately.

I'm pondering whether we should revert the PR, the original script does what it should do, my apologies. 
Wdyt?

rootzoll, 2024-12-19:
Yes I reverted back to the old script for now. It was quite some years back since I implemented it and not often testing it, so the details were also not present at my side. This was a good refresher.

PR #4886 (2024-12-18): Enhance BBcloseAllChannels.sh to allow LND user selection for channel closing method
fixes #4884 

Enhance BBcloseAllChannels.sh to allow user selection for channel closing method. 
Users can now choose between 'COOP' for cooperative close and 'FORCE' for forced close.
closed 2024-12-19 by rootzoll