raspiblitz/raspiblitz #4899: Stopped for manual provision while no audio device in Proxmox VM

adevoss, 2024-12-20:
After fresh install of Debian 12 and executing build_sdcard (`dev` branch) with option `-b v11-albyhub` the script finishes with the message:
REMOVE AUDIO DEVICE from VM before next boot up.

The VM configuration has no audio device (see image)
[image]

How do I manual provision?

rootzoll, 2024-12-20:
hmm .. this is normally just a trick for developers .. you have really no audio device added, strange.

Can you post here the result of command `lspci` from your booted up terminal?

adevoss, 2024-12-20:
There is an audio device ???? Didn't expect that.
`00:1b.0 Audio device: Intel Corporation 82801I (ICH9 Family) HD Audio Controller (rev 03)`
That's a question for the Proxmox forum then.

What script starts provisioning?

adevoss, 2024-12-20:
Not a Raspiblitz issue.
From the Proxmox forum: Machine `q35` has an audio device.

So I switched to default (`i440fx`). That has no audio device.
Now provisioning.

So if you use a VM (not Proxmox specific) doublecheck when installing if the VM has an audio device (`lspci`)

closed 2024-12-20 by adevoss