raspiblitz/raspiblitz #4901: Cannot recover node with lnd backup file

jetlagv1, 2024-12-23:
Since I moved my node to a raspi 5 with nvme drive it crashes every few weeks. Previous times I managed to recover but this time it ends with a failure. I start the recovery with a fresh SC card. Select new node, lnd and recover with lnd backup. It starts its process but ends with this failure message. So my question is, how do I recover this node now? 

########################################################### # /home/admin/raspiblitz.log

***********************************************             Running RaspiBlitz Bootstrap 1.11.3
Sun Nov  3 04:18:26 AM GMT 2024                             ***********************************************
  UNIT                      LOAD   ACTIVE SUB     DESCRIPTION
  avahi-daemon.service      loaded active running Avahi mDNS/DNS-SD Stack
  background.service        loaded active running RaspiBlitz Background Monitoring Service
  cron.service              loaded active running Regular background program processing daemon
  dbus.service              loaded active running D-Bus System Message Bus
  fail2ban.service          loaded active running Fail2Ban Service
  ModemManager.service      loaded active running Modem Manager
  NetworkManager.service    loaded active running Network Manager
  polkit.service            loaded active running Authorization Manager
  rsyslog.service           loaded active running System Logging Service
  smartmontools.service     loaded active running Self Monitoring and Reporting Technology (SMART) Daemon
  systemd-fsckd.service     loaded active running File System Check Daemon to report status
  systemd-journald.service  loaded active running Journal Service
  systemd-logind.service    loaded active running User Login Management
  systemd-rfkill.service    loaded active running Load/Save RF Kill Switch Status
  systemd-timesyncd.service loaded active running Network Time Synchronization
  systemd-udevd.service     loaded active running Rule-based Manager for Device Events and Files
  triggerhappy.service      loaded active running triggerhappy global hotkey daemon
  vnstat.service            loaded active running vnStat network traffic monitor

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
18 loaded units listed.
# make sure SSH server is configured & running
# *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
# countKeyFiles(6)
# DETECTED: SSHD NOT RUNNING --> Try reconfigure & kickstart again
INFO: _bootstrap.scan.sh loop started > sudo journalctl -f -u background.scan
# OK: SSHD RUNNING
## prepare raspiblitz temp
## INIT raspiblitz.info
state=starting
message=starting
setupPhase=boot
setupStep=0
baseimage=raspios_arm64
cpu=aarch64
vm=0
blitzapi=on
displayClass=lcd
displayType=
fsexpanded=0
btc_mainnet_sync_initial_done=0
btc_testnet_sync_initial_done=0
btc_signet_sync_initial_done=0
ln_lnd_mainnet_sync_initial_done=0
ln_lnd_testnet_sync_initial_done=0
ln_lnd_signet_sync_initial_done=0
ln_cl_mainnet_sync_initial_done=0
ln_cl_testnet_sync_initial_done=0
ln_cl_signet_sync_initial_done=0
# raspi_bootdir(/boot/firmware)
## redisEnabled(1)
## INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
- waiting for background.scan.service --> systemscan_runtime(1)
No Wifi config by file on sd card.
OK - logs are at 44 MB - within safety limit
Waiting for HDD/SSD ...
HDD/SSD connected: nvme0n1
No HDMI switch found.
No GPT issues detected on /dev/mmcblk0.
No GPT issues detected on /dev/nvme0n1.
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
[... 247 lines cut ...]
# OK - the bitcoin daemon on mainnet service is now enabled
# Add aliases bitcoin-cli, bitcoinlog
alias bitcoin-cli="sudo -u bitcoin /usr/local/bin/bitcoin-cli -rpcport=8332"
alias bitcoinlog="sudo -u bitcoin tail -n 30 -f /mnt/hdd/bitcoin/debug.log"
alias bitcoinconf="sudo nano /mnt/hdd/bitcoin/bitcoin.conf"
# OK - the bitcoindservice is enabled, to start manually use:
sudo systemctl start bitcoind
# Installed Bitcoin Core version v26.0.0

# Monitor the bitcoind with:
# sudo tail -f /mnt/hdd/bitcoin/debug.log

Provisioning bitcoin Testnet - not active
Provisioning bitcoin Signet - not active
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
##################
ERROR in _provision.setup.sh --> btc-testrun-fail on Mon Dec 23 01:34:02 AM GMT 2024\nDETAIL --> sudo -u bitcoin bitcoin-cli getblockchaininfo | grep initialblockdownload -c --> 0\nDEBUG --> /home/admin/raspiblitz.provision-setup.log
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(btc-testrun-fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
admin@10.0.0.64:~ ₿

rootzoll, 2025-01-04:
Your Bitcoind is not starting up anymore - most prob some corrupt blockchain data.

So what you could do is use command `repair` and use `RESET-CHAIN` to delete the blockchain and sync again. I know that comes with quite some wait but it makes sense to once try to sync fresh and if you then still run into crashes with your Raspi5+NVMe that it might be some hardware issue.

jetlagv1, 2025-01-05:
Re-downloading the blockchain and restoring with the backup recovered my node and channels.

closed 2025-02-03 by rootzoll