raspiblitz/raspiblitz #4911: v1.11.4 Password C multiple times

rootzoll, 2025-01-03:
looks like multiple reports that they have to enter password c multiple times until it works with lnd - check if lnd is not ready yet maybe

adevoss, 2025-01-03:
Just installed `dev` branch. Wanted to install v1.11.4 but there is no branch.
Same 'issue' here.

rootzoll, 2025-01-04:
> Wanted to install v1.11.4 but there is no branch.

There is the branch `v1.11` which contains the latest version if v1.11 which is v1.11.4

adevoss, 2025-01-05:
Installed branch `v1.11`. Result in

jalal, 2025-01-16:
Just to add -- I have the same problem. v1.11.1 works fine, upgrade to v1.11.4 gives the above problem. Downgrade to v1.11.1 and most things seem to work again. Not THUB or RTL though...

adevoss, 2025-01-17:
@jalal Do you use AlbyHub on `v1.11.1`? Requires manual installation < `v1.11.4`?
I'm trying to get AlbyHub to work on `v1.11.4` but can't determine if it is this issue or AlbyHub why it isn't working.

jalal, 2025-01-17:
> [@jalal](https://github.com/jalal) Do you use AlbyHub on `v1.11.1`? Requires manual installation < `v1.11.4`? I'm trying to get AlbyHub to work on `v1.11.4` but can't determine if it is this issue or AlbyHub why it isn't working.

No I don't. And THUB and RTL started working so needed time to settle back in to `v1.11.1`

scarolan, 2025-03-16:
Same issue here on latest.  Asks for password C several times.

asterysk, 2025-05-28:
I had same issue v1.11.4 on raspberry pi 4 8Gb. tried a fresh install without recovery, same problem (Lighning repeateddly asking for password ). Ended up doing another fresh install and selecting the alternate Lightning and no issues with password

henkisdabro, 2025-07-13:
Same issue here, keeps asking for Password C (on Raspberry Pi 8Gb). I hit `CTRL+C` and the `debug` and here are the logs if this helps. I'm nervous of making any attempts on my own, so will await instructions and assistance.

DEBUG LOGS:
[https://termbin.com/jyzs](https://termbin.com/jyzs)

rootzoll, 2025-07-30:
While testing on upcoming RaspiBlitz Verison v1.12.0 I have not seen the multiple Password C login so far.

closed 2025-07-30 by rootzoll