raspiblitz/raspiblitz #4914: WebUI/API hangs after initial Login

rootzoll, 2025-01-04:
This is translated from a the german forum:

```
Good morning everyone,  
Last week, I upgraded to the latest Raspiblitz version. Everything seemed to work fine at first, but unfortunately, when I access the WebUI via the IP address, I’m constantly stuck on a loading screen after logging in with Password A. This issue occurs on both the Home and Apps tabs.  

What can I do about this? A restart, for example, didn’t help. Interestingly, this problem wasn’t present immediately after the update but started appearing in the days following it.  

Thanks in advance for any help!  
It’s an unfortunately persistent and cross-version issue for which no real solution has been found yet. Sometimes deleting, uninstalling, or reinstalling the entire WebUI helps. Sometimes, `debug -l` can provide hints, for instance, whether nginx is reporting any errors.
```

Because we cannot reproduce the Problem on our test machines we would need some debug reports from the community when this error occurs. So if you see this problem with the WebUI, please login thru SSH and run `debug` on termonal and post the sections of API logs here.

adevoss, 2025-01-04:
Checked my node now after reading this issue.

- Installed the dev branch yesterday
- Now the Home page in the WebUI has the spinning circle
- Rebooted via Settings - Reboot
- Browser showed Login page while Raspiblitz was still rebooting
- Raspiblitz did not reboot so I stopped and started the VM
- Home page does not show Balance. It shows: -
- Apps page has the spinning circle

Output of debug command:
```
Printing debug logs. Be patient, this should take maximum 2 minutes ..
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory
```
and
https://termbin.com/22qk

rootzoll, 2025-01-04:
@adevoss thanks for the report .. your API in the logs shows OK, just a lof of `Jan 04 12:14:49 RASPIBLITZ python[948]: 2025-01-04 12:14:49 | ℹ️ | logging.py:87 | *.*.*.*:57024 - "GET /api/lightning/list-all-tx?reversed=true HTTP/1.0" 200` .. the Spinning Cirle on Apps menu was an issue here #3608 ... I reopened that one again for v1.12.0

Another report showed so far no problems with API in reported situatuion:
```
*** BLITZAPI STATUS ***
installed=1
repo='https://github.com/fusion44/blitz_api.git'
branch='blitz-v1.11'
commit='2bcdef1cef6718e13decd6cb01ef36b2345e51a4'
● blitzapi.service - BlitzBackendAPI
     Loaded: loaded (/etc/systemd/system/blitzapi.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-01-01 22:15:56 GMT; 2 days ago
   Main PID: 1053 (python)
      Tasks: 19 (limit: 9275)
        CPU: 1h 27min 54.834s
     CGroup: /system.slice/blitzapi.service
             └─1053 /home/blitzapi/blitz_api/venv/bin/python -m uvicorn app.main:app --port 11111 --host=*.*.*.* --root-path /api

Jan 04 11:05:37 raspberrypi python[1053]: 2025-01-04 11:05:37 | ℹ️ | logging.py:87 | *.*.*.*:50838 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 11:05:37 raspberrypi python[1053]: 2025-01-04 11:05:37 | ℹ️ | logging.py:87 | *.*.*.*:50840 - "POST /api/system/refresh-token HTTP/1.0" 200

*** LAST BLITZAPI LOGS ***
sudo journalctl -u blitzapi -b --no-pager -n20
Jan 04 10:14:01 raspberrypi sudo[2500520]: pam_unix(sudo:session): session closed for user ***
Jan 04 10:14:01 raspberrypi sudo[2500528]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/salt.txt
Jan 04 10:14:01 raspberrypi sudo[2500528]: pam_unix(sudo:session): session opened for user ***(uid=0) by (uid=117)
Jan 04 10:14:01 raspberrypi sudo[2500528]: pam_unix(sudo:session): session closed for user ***
Jan 04 10:14:02 raspberrypi sudo[2500740]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/a.hash
Jan 04 10:14:02 raspberrypi sudo[2500740]: pam_unix(sudo:session): session opened for user ***(uid=0) by (uid=117)
Jan 04 10:14:02 raspberrypi sudo[2500740]: pam_unix(sudo:session): session closed for user ***
Jan 04 10:14:02 raspberrypi python[1053]: 2025-01-04 10:14:02 | ℹ️ | logging.py:87 | *.*.*.*:35704 - "POST /api/system/login HTTP/1.0" 200
Jan 04 10:14:02 raspberrypi python[1053]: 2025-01-04 10:14:02 | ℹ️ | logging.py:87 | *.*.*.*:35722 - "GET /api/sse/subscribe HTTP/1.0" 200
Jan 04 10:14:03 raspberrypi python[1053]: 2025-01-04 10:14:03 | ℹ️ | logging.py:87 | *.*.*.*:35720 - "GET /api/lightning/list-all-tx?reversed=true HTTP/1.0" 200
Jan 04 10:14:23 raspberrypi python[1053]: 2025-01-04 10:14:22 | ℹ️ | logging.py:87 | *.*.*.*:43050 - "GET /api/lightning/list-all-tx?reversed=true HTTP/1.0" 200
Jan 04 10:14:43 raspberrypi python[1053]: 2025-01-04 10:14:43 | ℹ️ | logging.py:87 | *.*.*.*:50420 - "GET /api/lightning/list-all-tx?reversed=true HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44198 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44212 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44226 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44234 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44250 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 10:57:24 raspberrypi python[1053]: 2025-01-04 10:57:24 | ℹ️ | logging.py:87 | *.*.*.*:44264 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 11:05:37 raspberrypi python[1053]: 2025-01-04 11:05:37 | ℹ️ | logging.py:87 | *.*.*.*:50838 - "POST /api/system/refresh-token HTTP/1.0" 200
Jan 04 11:05:37 raspberrypi python[1053]: 2025-01-04 11:05:37 | ℹ️ | logging.py:87 | *.*.*.*:50840 - "POST /api/system/refresh-token HTTP/1.0" 200
```

Also direct API logs before and after login request dont show any specific problem:
[image]

@cstenglein as the API looks OK here, what could cause the spinning circle on login (any action after login hanging?) and what to ask people to check for to best debug?

adevoss, 2025-01-04:
My current Home page: (see image)
The Bitcoin Network is empty. Should this show Mainnet?
[image]

adevoss, 2025-01-05:
After installing v1.11.4 I had to reboot (spinning circles). Via WebUI does not work. Had to stop Proxmox VM. After reboot:
[image]
Second login attempt was successful. But now my Home page is:
[image]

adevoss, 2025-01-05:
Nothing done in the meantime. Now I have the spinning circle on Home and Apps

cstenglein, 2025-01-07:
Looks for me like a deeper problem.

If the app page does not load but the main page or parts of the main page are not shown, then the webui does not get all required messages from the blitz api. @rootzoll

@fusion44 is there logging implemented when getting the data for the SSE messages?

> Nothing done in the meantime. Now I have the spinning circle on Home and Apps
> Edit: And on AlbyHub

If it's on AlbyHub as well, then it seems to be some bitcoin core / lnd / web server issue?

Thunderhub also seems to have issues:

```
*** LAST 20 Thunderhub LOGS ***
sudo journalctl -u thunderhub -b --no-pager -n20
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   ],
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   level: 'error',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   message: 'Error connecting to node',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   timestamp: '2025-01-04T11:13:02.360Z'
Jan 04 12:13:02 RASPIBLITZ npm[7461]: }
Jan 04 12:13:02 RASPIBLITZ npm[7461]: {
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   message: 'Unable to connect to any node.',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   level: 'error',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   timestamp: '2025-01-04T11:13:02.368Z'
Jan 04 12:13:02 RASPIBLITZ npm[7461]: }
Jan 04 12:13:02 RASPIBLITZ npm[7461]: {
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   message: 'UnableToConnectToAnyNode',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   level: 'error',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   timestamp: '2025-01-04T11:13:02.370Z'
Jan 04 12:13:02 RASPIBLITZ npm[7461]: }
Jan 04 12:13:02 RASPIBLITZ npm[7461]: {
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   level: 'error',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   message: 'Initiating subscriptions failed: ',
Jan 04 12:13:02 RASPIBLITZ npm[7461]:   timestamp: '2025-01-04T11:13:02.371Z'
Jan 04 12:13:02 RASPIBLITZ npm[7461]: }
```

fusion44, 2025-01-07:
You should find a log message like this in the API logs:
`127.0.0.1:43276 - "GET /api/sse/subscribe HTTP/1.0" 200`
This at least indicates that the SSE connection was at least requested. 

@adevoss Could you please load the WebUI and ssh into the node and then take a look the the API logs a bit further to the past?
`sudo journalctl -u blitzapi -n 500 | grep -v "/api/lightning/list-all-tx?reversed=true"`
This will filter out the polls to list-all-tx.

Alternatively you can also follow the logs while reloading the page in your browser:
`sudo systemctl restart blitzapi && sudo journalctl -fu blitzapi`

I suspect that an app misbehaves and puts the API in an undefined state but the error message is buried a bit further down in the logs.

warioishere, 2025-01-07:
here some logs from a user which is also affected:

https://termbin.com/pjik

fusion44, 2025-01-08:
Ok thank you. We have some leads here, but I think two distinct problems.

@adevoss
Please try something. SSH into your VM and edit the following file:
`sudo nano /home/blitzapi/blitz_api/.env_sample`
Find the line `# platform=raspiblitz` and replace it with `platform=native_python`. Make sure to uncomment the line as the default is RasbpiBlitz when the line is commented out. Then restart the api service with `sudo systemctl restart blitzapi`. This will generate a new env file with the different platform. The Raspiblitz platform expects to be running on an actual PI. Platform Native Python may be more forgiving.

refs https://github.com/fusion44/blitz_api/issues/271

@warioishere 
This error is odd, but has nothing to do with the above problem. I have to take a closer look at the code in this area. Looks like we try to access a map key that is not existent. Now sure why this hasn't popped up somewhere else yet.

I've opened an issue on the blitz_api repo: https://github.com/fusion44/blitz_api/issues/277

warioishere, 2025-01-08:
> Ok thank you. We have some leads here, but I think two distinct problems.
> 
> @adevoss Please try something. SSH into your VM and edit the following file: `sudo nano /home/blitzapi/blitz_api/.env_sample` Find the line `# platform=raspiblitz` and replace it with `platform=native_python`. Make sure to uncomment the line as the default is RasbpiBlitz when the line is commented out. Then restart the api service with `sudo systemctl restart blitzapi`. This will generate a new env file with the different platform. The Raspiblitz platform expects to be running on an actual PI. Platform Native Python may be more forgiving.
> 
> refs [fusion44/blitz_api#271](https://github.com/fusion44/blitz_api/issues/271)
> 
> @warioishere This error is odd, but has nothing to do with the above problem. I have to take a closer look at the code in this area. Looks like we try to access a map key that is not existent. Now sure why this hasn't popped up somewhere else yet.
> 
> I've opened an issue on the blitz_api repo: [fusion44/blitz_api#277](https://github.com/fusion44/blitz_api/issues/277)

sorry i thought it might be a similar problem occuring because of the general hang but jst in a partiall situation.
The user jst sent me this picture with the logs I posted:

[image]

I am still learning to read code as good as possible so have some patients with me hehe :D

fusion44, 2025-01-08:
Don't worry, all good. Lets continue this conversation in the issue for this so we don't hijac this one: https://github.com/fusion44/blitz_api/issues/277

closed 2025-07-30 by rootzoll