raspiblitz/raspiblitz #4916: SSH connection refused

esmith413, 2025-01-04:
**Describe the bug**
A clear and concise description about the issue you encounter.

I can no longer ssh into raspiblitz. I always get a connection refused message.
**To Reproduce**
Steps to reproduce the behavior.
[image]

I was having issues with some of the applications not working and used ChatGPT to try and address them (checking channels using raspiblitz menu in command didn’t work, loop not working in RTL, among other things). Chatgpt somewhat buggered it up more so I decided I would start over. Backed up my LND and reflashed the SD card. After booting I was able to ssh in once, and then after that I get the connection refused message. 
-I have attempted adding an ssh file with no extension to the boot partition.
-I have turned off firewall and tried again with no luck.
-I have tried booting with an external monitor plugged in but I only get system status messages (pictures
 is attached) and the keyboard doesn’t do anything (using a Bluetooth usb for keyboard) so I can’t do anything with a command line.
-I was able to access the Web UI but now I’m stuck on waitsync with the note of provision done. I wasn’t able to backup my LND again so the backup is probably a day or so old.

I’m fine with starting over except that I have funds on the lightning network that I don’t want to lose. I went into the SSD and copied off the app-data folder and the lnd folder but I’m afraid to wipe the SSD and SD to start over in risk of losing the funds.

**Debug file**
If possible, please add a debug report. To do that, drag a text file with the debug report in here.

None.

adevoss, 2025-01-04:
> -I have attempted adding an ssh file with no extension to the boot partition.

That is a feature of PiOS not Raspiblitz. SSH is started on Raspiblitz.
Also use the correct password. After booting for the first time you will be asked to change the default password A.

> -I have turned off firewall and tried again with no luck.

What firewall did you turn off? If you directly connect to Raspiblitz there is no firewall involved (accept on the node itself but it is configured when you boot with a fresh sd-card.

> -I have tried booting with an external monitor plugged

Do you have a display on the Pi? When you boot for the first time you can monitor progress (instruction on the display).
If not keep the monitor plugged in while first booting the flashed sd-card.

> keyboard doesn’t do anything (using a Bluetooth usb for keyboard)

Don't know if BT is functioning. Please use a wired keyboard to be safe.

> I’m afraid to wipe the SSD

Please don't. All your data is there.
You can flash the sd-card with the OS as many times as you want.

Try flashing the sd-card again and boot with a display plugged in.
After first boot there is a recovery option.

esmith413, 2025-01-04:
> > -I have attempted adding an ssh file with no extension to the boot partition.
> 
> That is a feature of PiOS not Raspiblitz. SSH is started on Raspiblitz. Also use the correct password. After booting for the first time you will be asked to change the default password A.
> 
After first boot, I’m not able to get to that step because the ssh refuses the connection. The only way I get to setting password A and choosing the option to recover is through the Web UI. I’ve been using the same password each time. (I’ve reflashed the SD card several times). I never see the original blue screen in the commander prompt window like I did the first time I set up raspiblitz and was able to ssh.
> > -I have turned off firewall and tried again with no luck.
> 
> What firewall did you turn off? If you directly connect to Raspiblitz there is no firewall involved (accept on the node itself but it is configured when you boot with a fresh sd-card.
> 
I turned off the firewall on my laptop that is on the same local network.
> > -I have tried booting with an external monitor plugged
> 
> Do you have a display on the Pi? When you boot for the first time you can monitor progress (instruction on the display). If not keep the monitor plugged in while first booting the flashed sd-card.
> 
I have an external monitor plugged into the micro hdmi slot. I do not have the LCD screen setup for the Pi. When I boot the first time after reflashing the SD card I only see what I showed in the picture in the original message. There is no command line fore to enter anything.
> > keyboard doesn’t do anything (using a Bluetooth usb for keyboard)
> 
> Don't know if BT is functioning. Please use a wired keyboard to be safe.
> 
Ok I’ll try to find one. Is there a command to bring up a terminal from the system messages screen I see on the external monitor.
> > I’m afraid to wipe the SSD
> 
> Please don't. All your data is there. You can flash the sd-card with the OS as many times as you want.
> 
> Try flashing the sd-card again and boot with a display plugged in. After first boot there is a recovery option.
>
I’ve tried the recovery option a few times now (from the web UI) and the web UI stays on the waitsync status. The longest I let it go on this state was about 16 hours. Didn’t think it would need to do a full sync. But I’m not able to see anything else so not 100% sure if it’s hung up on something or not.

esmith413, 2025-01-04:
UPDATE:
I added a file named hdmi with no extension to the boot folder and the external monitor is working now. The Bluetooth keyboard is also working. The picture is where I’m at now.
[image]

if it hit Ctrl+C it just brings up a window saying raspiblitz is starting. What am I supposed to enter to enter commands?
Also during the boot I saw multiple fails in reference to ssh and a command I needed to run to get more info. I can’t remember what the command was but I think the error had something to do with OpenSBM?

adevoss, 2025-01-05:
> error had something to do with OpenSBM?
I guess it is OpenSSH (ssh daemon)

While booting for the first time you can monitor progress by watching raspiblitz.log in the admin home directory.
If ssh doesn't work you can take the sd-card to your laptop and see what is in the log files. Maybe there is a clue.

adevoss, 2025-01-05:
After first boot (with datadrive attached) this should be shown on the connected monitor.
[image]

adevoss, 2025-01-05:
Then this:
[image]

adevoss, 2025-01-05:
Then this:
[image]

adevoss, 2025-01-05:
When you press Ctrl+C you will enter a terminal. When you type the command you'll see the progress. It finishes (takes some time) with these messages.
[image]

adevoss, 2025-01-05:
Press Ctrl+C and `sudo reboot`
After reboot you will be asked for password C.
[image]

adevoss, 2025-01-05:
Then this will take a while. When ready the menu is shown.
[image]

esmith413, 2025-01-05:
Just an update:
To get to the terminal from the screen I showed, I had to press Ctrl + Alt + F2
From there I was able to find the things that were failing, one of which was ssh. I needed fix the permissions for the host keys.
I also had an error with bitcoind (thus the restarts) and lightning is waiting on bitcoin. Errors were a corrupt peers.dat file which I had created after moving the current one to *.bak and the chainstate was corrupted.
I believe I fixed the errors for bitcoind and now I’m waiting for -reindex to finish.

esmith413, 2025-01-05:
The ssh connection issue is fixed.
Plugged raspberry pi into external monitor with a hdmi cable plugged into micro hdmi slot and added a file named hdmi in the boot folder with no extension and booted the raspberry pi. Was able to login to raspiblitz and check the logs for ssh which told me that the permissions for the host keys were too loose. So I fixed the permissions with
sudo chown root:root /etc/ssh/ssh_host_*
sudo chmod 600 /etc/ssh/ssh_host_*
And restarted the service with
sudo systemctl restart ssh

rootzoll, 2025-02-03:
TODO: recheck all SSH cert scripts to set the correct permissions:
```
sudo chown root:root /etc/ssh/ssh_host_*
sudo chmod 600 /etc/ssh/ssh_host_*
```

rootzoll, 2025-05-15:
final testing PR #5001