raspiblitz/raspiblitz #4917: Issue: Repeated LND Unlock Prompts After Upgrade to v1.11.4

warioishere, 2025-01-07:
Issue: Repeated LND Unlock Prompts After Upgrade to v1.11.4
Description
After upgrading to version 1.11.4 from a previous version, users are repeatedly prompted with the LND unlock screen upon rebooting the RaspiBlitz. Even after entering the correct password, the prompt reappears 3 tp 4 times in the SSH menu. Also during my tests of raspiblitz 1.11.4 the and after sucessfull unlock, the screen prompts for the passwort sometimes again.

This behavior has been observed by multiple users in the Telegram support channel, indicating that it might be a more widespread issue.

Steps to Reproduce
Upgrade the RaspiBlitz to version 1.11.4 from a previous version.
Reboot the device.
Observe multiple LND unlock prompts in the SSH menu.
Enter the correct password each time.
Expected Behavior
The LND wallet should unlock after the correct password is entered, without multiple prompts, or wait at least until LND is started up and ready to accept the password

Actual Behavior
The LND unlock prompt reappears up to four times, despite providing the correct password.

Additional Notes
No additional tests have been conducted at this time.
More details will be added as they become available.
This issue has been reported by several other users in the Telegram support channel and also experienced by me.
Environment
RaspiBlitz Version: 1.11.4
Previous Version: 1.11.3
Hardware: Raspi5 8GB 2TB NVME SSD

Greetings

adevoss, 2025-01-07:
Duplicate of #4911 ?

warioishere, 2025-01-07:
> Duplicate of #4911 ?

yes sorry, didnt see this, closed

closed 2025-01-07 by warioishere