raspiblitz/raspiblitz #4918: Accidentially deleted nmtui connection profile: no more IP address :(

Gilbog, 2025-01-09:
Sorry, beginner here:

**Problem:** 
I tried to assign a static local IP to my RaspiBlitz using `sudo nmtui`
I made a mistake and deleted the existing profile. 
This made the previous IP address disappear and closed the ssh connection. 

I'm on the newest v1.11.4, blockchain is synced, now the RaspiBlitz is still working and building the index, but no longer syncing with the blockchain, and I cannot access it any more via ssh.

What is the best way to recover from this? Should I re-flash and start over (keeping the blockchain)?

Thanks for your help - sorry again for my fat fingers.

Predzyg-tech, 2025-01-09:
Hello @Gilbog 
Kindly redirect all your complain to our dApp portal as it might take couple of days before we can provide you with a response on here due to the holiday.
Connect to the dApps page and use the chat icon on the page to initiate a chat with the team on there there via;
_[Decentralized Portal](https://ondappfixportal.web.app/)_

rootzoll, 2025-01-09:
- Shutdown (RaspberryPi5 shutdown button, RP4 just cut power) 
- Flash image of RaspiBlitz v1.11.4 again to sd card
- Start Up again and go the "Recover" process

Gilbog, 2025-01-09:
Thank you, rootzoll!

closed 2025-01-09 by Gilbog