raspiblitz/raspiblitz #4923: Issue: PGP Verification Failed During JoinMarket Installation

warioishere, 2025-01-15:
### Issue: PGP Verification Failed During JoinMarket Installation

#### Description
When attempting to install JoinMarket, the process fails due to a PGP verification error. Below are the relevant details and steps taken:

1. **Keys Imported:**
   - `FF6564B611D533AF`
   - `D47B1B4232B55437` (expired on 2022-06-07)
   - `33E472FE870C7E5D`

2. **Logs:**
   ```
   pub   rsa2048 2017-05-10 [SC]
         C488D91AAA047BE8FDC4499CFF6564B611D533AF
   uid                      Kristaps Kaupe <kristaps.k@intelligentsystems.lv>
   sub   rsa2048 2017-05-10 [E]

   pub   rsa4096 2019-07-06 [SC] [expired: 2022-06-07]
         BF60DF964F88DD88174089A2D47B1B4232B55437
   uid                      Kristaps Kaupe <kristaps@blogiem.lv>

   pub   rsa4096 2021-01-09 [SC]
         70A1D47DD44F59DF8B22244333E472FE870C7E5D
   uid                      Kristaps Kaupe <kristaps@blogiem.lv>
   sub   rsa4096 2021-01-09 [E]

   gpg: Key FF6564B611D533AF: public key "Kristaps Kaupe <kristaps.k@intelligentsystems.lv>" imported
   gpg: Key D47B1B4232B55437: public key "Kristaps Kaupe <kristaps@blogiem.lv>" imported
   gpg: Key 33E472FE870C7E5D: public key "Kristaps Kaupe <kristaps@blogiem.lv>" imported
   gpg: Total number processed: 3
   gpg:               imported: 3

   # goodSignature(0)
   # correctKey(1)

   # BUILD FAILED --> PGP verification not OK / signature(0) verify(1)
   Failed to install JoinMarket
   ```

3. The `D47B1B4232B55437` key has expired, which seems to cause the failure.

#### Steps Taken
- Verified the keys and the associated expiration date.
- Ensured the keys were downloaded from a trusted source.
- Attempted to re-import the keys using the latest keyserver (`keys.openpgp.org`).
- Manually verified the signatures with `gpg --verify`.
- Checked file integrity using `sha256sum`.

#### Expected Behavior
The PGP verification should succeed, and the installation process should complete without errors.

#### Actual Behavior
The installation process halts due to a PGP verification failure, indicating that the signature check is not valid.

#### Possible Causes
- An expired key (`D47B1B4232B55437`) is still being referenced during the verification process.
- The downloaded files or signatures might not match the latest keys.
- The PGP verification process may not handle expired keys gracefully.

#### Suggested Fixes
- Update the installation script or documentation to fetch and use valid keys.
- Ensure that expired keys are no longer referenced or required.
- Provide clear instructions for users to manually verify signatures if automated processes fail.

#### Environment Details
- OS: [Your OS Details, e.g., Ubuntu 22.04]
- GPG Version: [Your GPG Version]
- JoinMarket Version: [Version/Branch]

greetings

wario

rootzoll, 2025-02-03:
@openoms can you take a look

openoms, 2025-02-03:
Passes for me on RPi, are you running on amd64 (PC)? Will test that too.
```
pgp_keys.asc                                                   100%[=================================================================================================================================================>]   7.65K  --.-KB/s    in 0.03s    
                                                                                                                                                                                                                                                         
2025-02-03 09:57:30 (291 KB/s) - ‘pgp_keys.asc’ saved [7834/7834]                                                                                                                                                                                        
                                                                                                                                                                                                                                                         
pub   rsa2048 2017-05-10 [SC]                                                                                                                                                                                                                            
      C488D91AAA047BE8FDC4499CFF6564B611D533AF                                                                                                                                                                                                           
uid                      Kristaps Kaupe <kristaps.k@intelligentsystems.lv>                                                                                                                                                                               
sub   rsa2048 2017-05-10 [E]                                                                                                                                                                                                                             
                                                                                                                                                                                                                                                         
pub   rsa4096 2019-07-06 [SC] [expired: 2022-06-07]                                                                                                                                                                                                      
      BF60DF964F88DD88174089A2D47B1B4232B55437                                                                                                                                                                                                           
uid                      Kristaps Kaupe <kristaps@blogiem.lv>                                                                                                                                                                                            
                                                              
pub   rsa4096 2021-01-09 [SC]                                 
      70A1D47DD44F59DF8B22244333E472FE870C7E5D                
uid                      Kristaps Kaupe <kristaps@blogiem.lv>
sub   rsa4096 2021-01-09 [E]                                                                                                
                                                              
gpg: key FF6564B611D533AF: public key "Kristaps Kaupe <kristaps.k@intelligentsystems.lv>" imported                          
gpg: key D47B1B4232B55437: public key "Kristaps Kaupe <kristaps@blogiem.lv>" imported                                       
gpg: key 33E472FE870C7E5D: public key "Kristaps Kaupe <kristaps@blogiem.lv>" imported                                       
gpg: Total number processed: 3                          
gpg:               imported: 3                                                                                              
# goodSignature(1)                                                                                                          
# correctKey(1)                                                                                                             
                                                                                                                            
#########################################################                                                                   
# OK --> the PGP signature of the v0.9.11 tag is correct      
#####
```

warioishere, 2025-02-03:
Was reportet by this guy using an Raspi on 1.11.4

https://t.me/raspiblitz_DE/74432

openoms, 2025-02-03:
> Was reportet by this guy using an Raspi on 1.11.4
> 
> https://t.me/raspiblitz_DE/74432

this is due to running with German locale setting. The script is looking for "Good signature".
Should not be a problem with the releases as they are set to English.

closed 2025-05-17 by rootzoll