raspiblitz/raspiblitz #4927: err="baseimage not supported"

stealth2600, 2025-01-17:
I attempted to update my USB flash drive to the latest prebuilt amd64 image of raspibilitz. Now when I reach the gnome desktop environment the system just reboots constantly. In my raspiblitz.log file I see the error  err="baseimage not supported" just before the reboots occur. I'm not sure how to resolve this. I've linked the prebuilt image I'm using below. Thanks for the help!

https://github.com/raspiblitz/raspiblitz/actions/runs/12162657858

openoms, 2025-01-18:
Please describe the exact setup. What is the hardware you are using (computer, disk sizes, what is the size of the USB drive)?

Paste the output of your disk setup with:
`lsblk`
and
`df -h`

The reboot is cause by one of the background services, let's disable that first.

openoms, 2025-01-18:
Likely the reboots are because of this code part:
https://github.com/raspiblitz/raspiblitz/blob/2a30efdbbb57a03121b647e2cef225819d719aac/home.admin/_bootstrap.sh#L489-L514

You could try commenting it out on your OS disk at:
`/home/admin/_bootstrap.sh`
to stop the bootloop.

stealth2600, 2025-01-18:
> Please describe the exact setup. What is the hardware you are using (computer, disk sizes, what is the size of the USB drive)?
> 
> Paste the output of your disk setup with: `lsblk` and `df -h`
> 
> The reboot is caused by one of the background services, let's disable that first.

The setup is a mini pc with Ryzen 7 5825U processor, 32gb DDR4, 128GB NVME drive In an enclosure that's connected to PC via USB-C (system drive) and a 2TB NVME that's internal (data drive). There's also an 8gb USB Flash Drive that I'm using for backups.

admin@192.168.2.9:~ ₿ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1   7.5G  0 disk 
└─sda1        8:1    1   7.5G  0 part /mnt/backup
                                      /media/admin/BLITZBACKUP
sdb           8:16   0 119.2G  0 disk 
├─sdb1        8:17   0   512M  0 part /boot/efi
└─sdb2        8:18   0 117.8G  0 part /
zram0       253:0    0     0B  0 disk 
nvme0n1     259:0    0   1.9T  0 disk 
└─nvme0n1p1 259:1    0   1.9T  0 part /mnt/hdd

admin@192.168.2.9:~ ₿ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             15G     0   15G   0% /dev
tmpfs           2.9G  1.7M  2.9G   1% /run
/dev/sdb2       116G   14G   96G  13% /
tmpfs            15G     0   15G   0% /dev/shm
tmpfs           5.0M   12K  5.0M   1% /run/lock
tmpfs            32M  4.0K   32M   1% /var/cache/raspiblitz
/dev/sdb1       511M  5.9M  506M   2% /boot/efi
/dev/nvme0n1p1  1.9T  797G  986G  45% /mnt/hdd
tmpfs           2.9G  2.5M  2.9G   1% /run/user/1001
/dev/sda1       7.5G  1.1M  7.5G   1% /mnt/backup

stealth2600, 2025-01-18:
Thanks for the recommendations. I was able to stop the reboot loops myself before I saw your comments. I did this by commenting out lines 510-512 in _bootstrap.h. Next, I had to update my bitcoin core to version 18.0 and lnd to version 0.18.4-beta and this allowed the provisioning to work properly. I did run into an issue with mempool, something with the mariadb-server didn't work correctly and I'm still trying to sort that out. I did try using the bonus.mempool.sh script, but that didn't resolve the issue. Otherwise, my node is back online and operational.

I tried the command:

sudo /home/admin/config.scripts/blitz.datadrive.sh uasp-fix

I received the same output as you, so I don't think this was the cause of my troubles.

Please let me know what other information I can provide or test I can run to help you get to the root cause. Thanks again for all your help and support!

stealth2600, 2025-01-18:
@openoms Here are the errors I'm seeing when trying to install mempool. Its hanging on the Setting up mariadb-server step. 

sudo journalctl --no-pager -u mariadb.service

Jan 18 04:52:57 digitalparadox systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
Jan 18 04:52:57 digitalparadox mariadbd[1354]: 2025-01-18  4:52:57 0 [Note] Starting MariaDB 10.11.6-MariaDB-0+deb12u1 source revision  as process 1354
Jan 18 04:52:57 digitalparadox mariadbd[1354]: 2025-01-18  4:52:57 0 [ERROR] mariadbd: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [ERROR] mariadbd: Got error 'Could not get an exclusive lock; file is probably in use by another process' when trying to use aria control file '/var/lib/mysql/aria_log_control'
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [Note] InnoDB: The first data file './ibdata1' did not exist. A new tablespace will be created!
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [Note] InnoDB: Compressed tables use zlib 1.2.13
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [Note] InnoDB: Number of transaction pools: 1
Jan 18 04:53:27 digitalparadox mariadbd[1354]: 2025-01-18  4:53:27 0 [Note] InnoDB: Using crc32 + pclmulqdq instructions

stealth2600, 2025-01-19:
@openoms I've complete removed mariadb and mysql and reinstalled everything. I am now unable to start mariadb and I get the following error. Can you provide any help or advice?

Jan 19 12:23:59 digitalparadox systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
Jan 19 12:23:59 digitalparadox sh[1106]: /usr/bin/galera_recovery: 1: /usr/bin/my_print_defaults: not found
Jan 19 12:23:59 digitalparadox (mariadbd)[1114]: mariadb.service: Failed to locate executable /usr/sbin/mariadbd: No such file or directory
Jan 19 12:23:59 digitalparadox (mariadbd)[1114]: mariadb.service: Failed at step EXEC spawning /usr/sbin/mariadbd: No such file or directory
Jan 19 12:23:59 digitalparadox systemd[1]: mariadb.service: Main process exited, code=exited, status=203/EXEC
Jan 19 12:23:59 digitalparadox systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 19 12:23:59 digitalparadox systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.

openoms, 2025-01-19:
Mariadb is installed like this:

https://github.com/raspiblitz/raspiblitz/blob/61766d8beaa1641e59d40c693a1c40d5d2ff9548/home.admin/config.scripts/bonus.mempool.sh#L132

Try to uninstall mempool then remove mariadb completely to start clean as possible:

```
config.scripts/bonus.mempool.sh off
config.scripts/bonus.mempool.sh uninstall

sudo apt-get remove -y mariadb* --purge

config.scripts/bonus.mempool.sh on
```

stealth2600, 2025-01-19:
@openoms thanks for the advice/instructions. I followed them exactly and I've been stuck at 

Setting up mariadb-server (1:10.11.6-0+deb12u1) ...

For about an hour now. I can't seem to find any useful logs, but I do see a bunch of processes running related to the mariondb-server install. Please see below. How long should this take? It feels like it shouldn't take this long and I'm not sure where to go from here. Any advice you can offer is much appreciated, I have no idea what's going on.\

admin@192.168.2.9:~ ₿ ps aux | grep mysql
root       97582  0.0  0.0   7384  3376 pts/4    S+   16:46   0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root       97605  0.0  0.0   7384  2016 pts/4    S+   16:46   0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root       97606  0.0  0.0   7384  1852 pts/4    S+   16:46   0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root       97607  0.0  0.0   7384  2016 pts/4    S+   16:46   0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root       97608  0.0  0.0   5924  1716 pts/4    S+   16:46   0:00 cat /usr/share/mysql/mysql_system_tables.sql /usr/share/mysql/mysql_performance_tables.sql /usr/share/mysql/mysql_system_tables_data.sql /usr/share/mysql/fill_help_tables.sql /usr/share/mysql/maria_add_gis_sp_bootstrap.sql /usr/share/mysql/mysql_sys_schema.sql
mysql      97610  0.0  0.2 462568 70212 pts/4    Dl+  16:46   0:00 /usr/sbin/mariadbd --lc-messages-dir=/usr/share/mysql/english/.. --bootstrap --silent-startup --basedir=/usr --datadir=/var/lib/mysql --log-warnings=0 --enforce-storage-engine= --plugin-dir=/usr/lib/mysql/plugin --disable-log-bin --user=mysql --max_allowed_packet=8M --net_buffer_length=16K

stealth2600, 2025-01-20:
Since we've diverged into a new problem related to mempool / mariadb-server installation, I've created the issue linked below. I'm going to close this issue at this time since I was able to sneak by this reboot loop issue by commenting out some lines in _bootstrap.sh as suggested by @openoms 

https://github.com/raspiblitz/raspiblitz/issues/4937

closed 2025-01-20 by stealth2600