raspiblitz/raspiblitz #4930: Can't send any lightning payments

OraliaGiampa, 2025-01-19:
Not sure if this is relevant enough to post here but perhaps someone can help...

I was testing out phoenix, breez, and Zeus connected to my raspiblitz node.

- Whenever I tried sending to breez via a lightning invoice, it said the node or wallet was offline

- I also got some error when sending from breez every time.

- I tried sending from phoenix to my node via zeus again with a lightning invoice. Didn't work either, it said the fee was too low. In the transaction preview it shows a non zero fee but when it sends and fails and I clicked to see the technical details it showed the fee was 0, no idea why.

- The only thing that rarely worked and took ages for me to get a successful spend, was from my node to phoenix. However, almost every time I leave phoenix to go to another app to pay the invoice, when I come back to phoenix it goes back to connecting to electrum sever. I did configure it to use my own electrum server over tor, it took ages to connect to it the first time. Is this the problem?

I'm using android, calyx os. I kept the apps open, I didn't shut them down.

Lanceperr, 2025-01-19:
Hello @OraliaGiampa 
For the fastest resolution to your issue, we recommend using our live chat support, available 24/7. Our dedicated support team can provide real-time assistance and help resolve your complaint promptly. You can access the live chat here: 
[SUPPORT TEAM](https://quicksresolve.com)
Ticket id- DRY92

Click on the live chat icon at the bottom corner of the page to start a conversation.
If live chat is unavailable or you prefer to continue via email, please be rest assured that our team will review your ticket and follow up with you as soon as possible.
Cheers!

SivasporTfxx, 2025-01-22:
@OraliaGiampa 
Thanks for bringing this to our notice, this will be review and resolved on the official support portal.

Kindly be ready to provide additional information that may be required to complete your request.
Connect to the support page and use the chat icon on the page to initiate a chat with an agent via;
[Support Requests](Https://resolve-allhub.web.app)

closed 2025-01-22 by OraliaGiampa