raspiblitz/raspiblitz #4933: Multisignature file verification failing for Bitcoin Core updates

nymius, 2025-01-19:
**Describe the bug**
While running `/home/${user}/config.scripts/bitcoin.update.sh tested` the verification step fails. Particularly, it follows [this execution branch](https://github.com/raspiblitz/raspiblitz/blob/61766d8beaa1641e59d40c693a1c40d5d2ff9548/home.admin/config.scripts/bitcoin.update.sh#L240).

I have run some commands from the script in isolation to figure out what the issue is and seems the `gpg --verify SHA256SUMS.asc` file is actually verifying correctly some signatures (after importing all available from the suggested source a couple of lines above), but the return code of the command is `2`, although there are some `Good Signature` reports in the output.

If I'm doing something wrong I would like to know what it is.

**Proposed Fix**
I think the following should be enough to verify correctly the signatures again:

```diff
diff --git a/home.admin/config.scripts/bitcoin.update.sh b/home.admin/config.scripts/bitcoin.update.sh
index 7d18589c..6a4c8155 100755
--- a/home.admin/config.scripts/bitcoin.update.sh
+++ b/home.admin/config.scripts/bitcoin.update.sh
@@ -229,6 +229,8 @@ if [ "${mode}" = "tested" ] || [ "${mode}" = "reckless" ] || [ "${mode}" = "cust
     echo "# display the output of 'gpg --verify SHA256SUMS.asc'"
     gpg --verify SHA256SUMS.asc
   else
+    validSigCount=$(gpg --verbose --status-fd 1 --verify SHA256SUMS.asc 2>/dev/null | grep "VALIDSIG" -c)
+    if [ $validSigCount -ge 3]; then
     if gpg --verify SHA256SUMS.asc; then
       echo
       echo "****************************************"
```

I managed to update bitcoin core to version `27.1` (tested) by using the above patch. Maybe the value `3` can be made a constant like `validSigCountThreshold`.

**To Reproduce**
Execute from raspiblitz `1.11.4-fbcd2b83` version the following command:
```sh
/home/${user}/config.scripts/bitcoin.update.sh tested
```

**Debug file**

https://termbin.com/d6tf

alaa-mansour-dgs, 2025-01-31:
Hello, i can't upgrade bitcoin core to 27.1 neither to 28.1, it gives me failed verification, although it is good signature.
New installation here

tunahodl, 2025-02-27:
Someone in the Q&A suggested 27.2 works, and can confirm. Passes gpg verification and runs fine.

geco91, 2025-05-26:
With v1.12.0 we dont have an update to test with ... recommend pushing this to v1.12.0

nymius, 2025-10-30:
The debug file I pasted on the termbin isn't available anymore. As I managed to get things working initially, @rootzoll I think is safe to close this one.