raspiblitz/raspiblitz #4936: mempool.space causes high CPU load after upgrade to 1.11.4

woiwoiwoiwoi, 2025-01-20:
**Describe the bug**
I'm experiencing high CPU load after upgrade to 1.11.4. On 1.11.2, average CPU load was 2-2.5. After the update, it was around 8 for a day. Now it's dropped to around 4.4. On a four-core CPU (RPi 4, 8GB) that's still too high, isn't it?

Deactivating mempool instantly drops load to 1.5. After re-enabling mempool, average load jumps to 4+ again.

As far as I can tell, the tx index is built. At least the info screen doesn't mention anything and I can use mempool without issues.

**To Reproduce**

Don't know if it's relevant, but after the upgrade, mempool did not work. Deactivating and reactivating did not help. I had to run:
```
/home/admin/config.scripts/bonus.mempool.sh off
/home/admin/config.scripts/bonus.mempool.sh uninstall
/home/admin/config.scripts/bonus.mempool.sh on
```

The only service I added after the upgrade was LND, later Alby Hub. No channels open yet though. And I'm pretty sure CPU load was high before enabling LND.

woiwoiwoiwoi, 2025-02-01:
For the past two weeks, CPU load was constantly 4-4.5. The other day I then installed Specter Desktop and then turned it off again because it did not work (which is a different issue, I guess). Now CPU load is between 2 and 3, sometimes as low as 1.7.

This all seems really weird to me. Or is there some logic to it that I don't see?

rootzoll, 2025-02-03:
Just tested mempool on Raspi5 again .. having quite low CPU use.

TODO: Test mempool on Raspi4

woiwoiwoiwoi, 2025-03-02:
Following the recommendation on the Telegram channel, I updated LND to 0.18.5. Afterwards, CPU load was constantly between 4-4.5. After 10 days, it dropped to around 2.3. Is that expected? Does LND cause high CPU load for a long-ish period?

For testing purposes, I then disabled mempool via the ssh menu and load dropped to 1.7-2. I only left it for a few minutes and turned it back on. Load is back to 3.5-4. Will maybe take a bit to stabilise again.

But more generally, is a load of 4.5 an issue or is it not problematic?

rootzoll, 2025-05-17:
LND 0.18.5 or higher will be part of RaspiBlitz 1.12.0 .. also checking for new mempool releases, maybe they did some optimization in th meantime.

PR #5012 (2025-05-19): UPdate Mempool Space to 3.2.1
closes #4936
closed 2025-05-19 by rootzoll