raspiblitz/raspiblitz #4937: AMD64 Issue: Mempool fails to install on Debian 12

stealth2600, 2025-01-20:
@openoms Per our discussion on telegram, I've created a new issue for this problem.

**Describe the bug**
I am unable to install mempool on a Debian 12 system because mariadb-server fails to install properly. The installation simply hangs and all related processes go to a sleep state. I let the install sit for over 24 hours with no change in state. I could not find any logs detailing an error. Debian 12 install mariadb-server 1:10:11.6-0+deb12u1 from the main Debian repository.

**To Reproduce**
I built my Debian 12 system using the debian-12.9.0-amd64-DVD-1.iso file which I flashed to a USB drive. I then attached my "OS drive " which is an NVME drive that's in an enclosure and attached to the computer via a usb-c cord. I also connect the Debian 12 USB install drive to my computer and proceeded to install the OS following the steps in the link below.

https://github.com/raspiblitz/raspiblitz/blob/dev/alternative.platforms/README.md#install-debian-to-the-os-disk

When the OS completed its installation, I rebooted the system and continued with the installation steps as per below.

https://github.com/raspiblitz/raspiblitz/blob/dev/alternative.platforms/README.md#install-the-raspiblitz-scripts

After finishing the raspiblitz install, I shutdown my system and installed my "Data Drive", which is a 2 TB internal NVME drive. The drive already had my node data on it from my previous installation. I did have an issue with raspiblitz causing a reboot loop on the initial startup, but I was able to overcome that by commenting out some lines in _bootstrap.sh. Please see the issue linked below for more details. The root cause of the reboot loop was never identified.

https://github.com/raspiblitz/raspiblitz/issues/4927

Once I stopped the reboot loop, raspiblitz correctly identified the old installation on my data drive and proceeded with the provisioning steps. After completing the provisioning, I rebooted my system and verified that bitcoin core and lnd were synced up and working correctly. I also checked on electrs and public-pool since I am also using those services and they worked fine too. I then went ahead and attempted the mempool install. I tried installing from the raspiblitz menu and also with the bonus.mempool.sh script, but the installation of mariadb-server installation hung no matter how I attempted the install. 

Setting up mariadb-server (1:10.11.6-0+deb12u1) ...

I let my system sit at this step for over 24 hours and it didn't proceed forward. In addition, I could see the processes below were all just sitting in a sleep state. I have not been able to determine what these processes are waiting for using strace. 

admin@192.168.2.9:~ ₿ ps aux | grep mysql
root 97582 0.0 0.0 7384 3376 pts/4 S+ 16:46 0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root 97605 0.0 0.0 7384 2016 pts/4 S+ 16:46 0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root 97606 0.0 0.0 7384 1852 pts/4 S+ 16:46 0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root 97607 0.0 0.0 7384 2016 pts/4 S+ 16:46 0:00 bash /usr/bin/mysql_install_db --rpm --cross-bootstrap --user=mysql --disable-log-bin --skip-test-db
root 97608 0.0 0.0 5924 1716 pts/4 S+ 16:46 0:00 cat /usr/share/mysql/mysql_system_tables.sql /usr/share/mysql/mysql_performance_tables.sql /usr/share/mysql/mysql_system_tables_data.sql /usr/share/mysql/fill_help_tables.sql /usr/share/mysql/maria_add_gis_sp_bootstrap.sql /usr/share/mysql/mysql_sys_schema.sql
mysql 97610 0.0 0.2 462568 70212 pts/4 Dl+ 16:46 0:00 /usr/sbin/mariadbd --lc-messages-dir=/usr/share/mysql/english/.. --bootstrap --silent-startup --basedir=/usr --datadir=/var/lib/mysql --log-warnings=0 --enforce-storage-engine= --plugin-dir=/usr/lib/mysql/plugin --disable-log-bin --user=mysql --max_allowed_packet=8M --net_buffer_length=16K

Currently my system is operating correctly, with the exception of being unable to install mempool due to this mariadb-server installation issue. I've included a debug.log file below for your reference. I was playing around with electrs, so you can ignore those reference in the debug log, they have nothing to do with the issue highlighted above.

Any help, advice or recommendation for how to resolve this would be greatly appreciated. 

If possible, please add a debug report. To do that, drag a text file with the debug report in here.
[debug logs.txt](https://github.com/user-attachments/files/18482765/debug.logs.txt)

wchancao, 2025-01-21:
Yes, something is wrong with installing mempool space in raspiblitz v1.11.4 running on debian 12... these are my results...

```
admin@192.168.0.7:~ ₿ config.scripts/bonus.mempool.sh on
# making sure services are not running
# *** Activate MEMPOOL ***
# txindex is already active
# getting RPC credentials from the bitcoin.conf
sending incremental file list
./
browser/
browser/ar/
browser/ar/145.4bb5b3cfab1690c1.js
browser/ar/182.2e411f53a29cab93.js
browser/ar/208.e6088b4e2ef417c8.js
browser/ar/250.fbc84453801fcda5.js
browser/ar/26.334002e82dd4d358.js
browser/ar/368.2a5f2dfaf81fc38d.js
browser/ar/371.b22ad4ca4ef9d205.js
browser/ar/3rdpartylicenses.txt
browser/ar/428.fcc2ba9aa78dabff.js
browser/ar/43.53a4eef709e1034d.js
browser/ar/456.e6fe01cfe4731814.js
browser/ar/463.5787601d0ff996c1.js
browser/ar/492.2ecf2392f6b0e820.js
browser/ar/54.110b36cede720439.js
browser/ar/559.c6407603b0cb8156.js
browser/ar/576.649b9d2fe929f164.js
browser/ar/61.5eb6009d210c134a.js
browser/ar/635.2310155448866c8c.js
browser/ar/647.9dcc6fc8c4d1316f.js
browser/ar/687.a5415e0850faf46b.js
browser/ar/807.09ce51e60af47dec.js
browser/ar/815.84ef33ed40a524ad.js
browser/ar/836.1d75cbde07161e4f.js
browser/ar/881.baa13afeb716cd74.js
browser/ar/892.3bfd31e22474b692.js
browser/ar/924.08b9c5aa534a326c.js
browser/ar/946.a1a0e6150b320625.js
browser/ar/973.2e299e277defa8fe.js
browser/ar/974.81afc301037ba643.js
browser/ar/976.3371c60e68f41a50.js
browser/ar/981.986ed6f7c071acc5.js
browser/ar/989.acd179c19a36c7e3.js
browser/ar/bukele.css
browser/ar/contrast.css
browser/ar/favicon.ico
browser/ar/index.html
browser/ar/main.8e270f30ac27ea31.js
browser/ar/polyfills.c41df647371e42d1.js
browser/ar/robots.txt
browser/ar/runtime.7f4fb53c79a686e6.js
browser/ar/styles.bd4eb3a87fd6de43.css
browser/ar/wiz.css
browser/ca/
browser/ca/145.4bb5b3cfab1690c1.js
browser/ca/182.2e411f53a29cab93.js
browser/ca/208.e6088b4e2ef417c8.js
browser/ca/250.fbc84453801fcda5.js
browser/ca/26.334002e82dd4d358.js
browser/ca/368.2a5f2dfaf81fc38d.js
browser/ca/371.b22ad4ca4ef9d205.js
browser/ca/3rdpartylicenses.txt
browser/ca/428.fcc2ba9aa78dabff.js
browser/ca/43.53a4eef709e1034d.js
browser/ca/456.e6fe01cfe4731814.js
browser/ca/463.5787601d0ff996c1.js
browser/ca/492.2ecf2392f6b0e820.js
browser/ca/54.110b36cede720439.js
browser/ca/559.c6407603b0cb8156.js
[... 1395 lines cut ...]
browser/zh/favicon.ico
browser/zh/index.html
browser/zh/main.8e270f30ac27ea31.js
browser/zh/polyfills.c41df647371e42d1.js
browser/zh/robots.txt
browser/zh/runtime.7f4fb53c79a686e6.js
browser/zh/styles.bd4eb3a87fd6de43.css
browser/zh/wiz.css

sent 154,037,982 bytes  received 27,471 bytes  308,130,906.00 bytes/sec
total size is 153,890,146  speedup is 1.00
# *** Updating Firewall ***
Rule updated
Rule updated (v6)
Rule updated
Rule updated (v6)

nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
*** Install mempool systemd ***
Created symlink /etc/systemd/system/multi-user.target.wants/mempool.service → /etc/systemd/system/mempool.service.
# OK - the mempool service is now enabled
# OK - the mempool.service is enabled, system is on ready so starting service
result='mempool service not active'
```

stealth2600, 2025-01-21:
I noticed that zram is also not working properly on my system. In an effort to resolve this, I tried to install zram-tools and it is also hung just like mariadb-server. Something isn't right here, I'm not sure why this is happening for certain packages.

stealth2600, 2025-01-21:
I can also confirm this is NOT Debian 12 issue. I was able to successfully install mariadb-server, mariadb-client and zram-tools on a fresh Debian 12 install BEFORE running the build_secard.sh script.

stealth2600, 2025-01-21:
I've manages to get my system fully operational. I did two things differently this time. I used the debian-12.9.0-amd64-netinst.iso when creating my "OS Drive" instead of the DVD image. In addition, I installed mariadb-server, mariadb-client and zram-tools before I ran the build_sdcard.sh script on my freshly installed Debian 12 system. I'm not sure which of these two actions solved the problem for me, but I'm leaning towards the usage of the debian-12.9.0-amd64-netinst.iso. I say this because it automatically selects the right repositories since you are using them to build the OS. The DVD iso on the other hand doesn't choose the appropriate repositories and it's left up to the user to select them. Being new at running my Debian 12, I likely chose the wrong repositories and this caused my 3 day headache. I will say that I still did encounter a reboot loop after booting the system for the first time, but that was easily resolved by commenting out the necessary lines in the _bootstrap.sh script in order to allow the provisioning to execute.

adevoss, 2025-01-21:
> The DVD iso on the other hand doesn't choose the appropriate repositories 

What do you mean by this, can you explain?
The repository mirror you select when installing Debian are all the same accept for location (country).

For installing Raspiblitz I use the 12.0.0 DVD iso file (I'm too lazy to download every point release) on Proxmox.
When installing Debian on a VM I choose a mirror near my location but I think every mirror will work (maybe slower because of distance).
When installing I **deselect** any desktop environment and **select** ssh server.

After that is finished (and reboot because of newer kernel) Debian is up to date. No problems running `build_sdcard` (`-d headless -w off -t 0`).
Both my disks (OS and datadrive) are virtual. No physical disks attached to server just for Raspiblitz.

ii  mariadb-server                       1:10.11.6-0+deb12u1
Mempool is OK after installing `v1.11.4`.

adevoss, 2025-01-21:
Isn't this issue solved now you know you have to select a mirror?

I don't know if zram is supported by raspiblitz.
Have you tested zram on Debian 12 without raspiblitz?

stealth2600, 2025-01-21:
@adevoss yes, the issue is resolved. There is some sort of a reboot loop issue with the _bootstrap.sh script, but as state I was able to overcome that by commenting out some lines of code in the script. ZRAM is working correctly on my Debian 12 system too. I think the initial issue with zram was caused by the same mirror/repository issue previously mentioned. I'll close the issue now, thanks for chiming in.

closed 2025-01-21 by stealth2600

adevoss, 2025-01-21:
For the reboot issue there is #4929 I think. You can try again after that PR is merged.