raspiblitz/raspiblitz #4943: Update: Alby Hub --> 1.13, .14

itstomekk, 2025-01-28:
Hey, following [this ](https://github.com/raspiblitz/raspiblitz/issues/4727#issuecomment-2561880042)comment, I'm letting know, that there is already a new release of [Alby Hub](https://github.com/getAlby/hub/releases/), 1.13

It comes with signatures as requested (and needed). I believe there is no needed changes in config, but that needs confirmation from @rolznz , who might be Alby's named dev contact, as @rootzoll has asked.

Probably by the next milestone of Raspiblitz, there will be another release of the Hub, so let's wait till then with the update?

Let me know if I can help further

rolznz, 2025-01-29:
@itstomekk no changes in the config are needed as far as I know.

Note: the title also says, .14. 1.14.0 is not released yet, but hopefully soon.

itstomekk, 2025-01-29:
Yes, that's why I put .14 to be aware that new release is coming out soon, and probably there will be another one before Raspiblitz releases. Thanks Roland!

rootzoll, 2025-05-17:
RaspiBlitz v1.12.0 will come with latest v1.17.0 Alby Hub Closing this issue.

closed 2025-05-17 by rootzoll