raspiblitz/raspiblitz #4945: Blockchain and Lightning 100.00% but shows "still syncing" on ssh and "Not synced" in Thunderhub

bfeeny, 2025-01-30:
**Describe the bug**
I had a working 1.7.1 node.  Upgraded to 1.11.4.

LCD Screen shows normal lightning bolt info screen and shows Sync 100%
When I go into Thunderhub it shows "Not Synced" in red on the upper left of the main screen
In RTL it shows red circle next to Bitcoin Mainnet and "Not Synced to Chain"
On the SSH screen it shows Blockchain Progress 100.00% LIghtning Progress 100.00% but at the top says "Node is Syncing"

**To Reproduce**
Working 1.7.1 node upgradeed to 1.11.4

**Debug file**

https://termbin.com/cni5

rootzoll, 2025-02-03:
Hi. Thats quite a big jump in versions we never tested.

Does it show with bitcoin & lnd peers greater >0 and is it still the same after 4 days?

bfeeny, 2025-02-03:
It looks like after time, it has fully synced both blockchain and LND.  The inconsistencies in where it said synced or not synced is likey some sort of bug.  Here is current state:

https://termbin.com/qm4t