raspiblitz/raspiblitz #4951: After upgrade to 1.11.4, Specter Desktop does not work

woiwoiwoiwoi, 2025-02-03:
**Describe the bug**
I updgraded from 1.11.2 (which was a fresh install) to 1.11.4. I then used the ssh menu to install Specter Desktop. The install went fine, but going to the URL:port did not load any page. I then uninstalled it again via ssh menu.

Interestingly, I had been experiencing [high CPU load after the upgrade to 1.11.4, apparently due to mempool.space](https://github.com/raspiblitz/raspiblitz/issues/4936). After installing and uninstalling Specter Desktop, load has dropped significantly. I don't know if this can even be related, but I thought I'd mention it here.

**To Reproduce**

1. install 1.11.2 fresh
2. enable Blitz API + WebUI, Electrs, mempool.space, and RPC Explorer
3. upgrade to 1.11.4
4. issues with mempool.space > uninstall it via script > reinstall it
5. enable LND and Alby Hub via ssh menu
6. enable Specter Desktop via ssh menu
7. go to URL:port > nothing loads

rootzoll, 2025-05-17:
Testing Specter for v1.12.0 release ... there is general question:

The latest version 2.1.1 (released beginning of the year) says: 
```
Docker Only Release - For this release there are only docker images available. The easiest way to run Specter via a docker image is with a node solution like Umbrel.
```
https://github.com/cryptoadvance/specter-desktop/releases/tag/v2.1.1 

And the over a year version 2.0.5 before that says:
```
Please create a full backup before migrating or any major internal changes like switching to an electrum based installation. You can easily create a backup in Settings --> Backup Specter (zip file).
```
https://github.com/cryptoadvance/specter-desktop/releases/tag/v2.0.5

@k9ert raspiblitz is not planning to run docker containers in the future - so does it make sense to wait for a new update or should be offer user a migration?

k9ert, 2025-05-20:
@moneymanolis any idea here?

moneymanolis, 2025-05-23:
IMO there won't be a complete binary release of Specter anytime soon.