raspiblitz/raspiblitz #4966: Retest "headless" build script

rootzoll, 2025-02-26:
There was a user report running the build script started with "lcd=headless" produced "displayClass=lcd" in raspiblitz setup. 

- [ ] try to reproduce

fanaticalfishing, 2025-03-28:
It seems to work OK for me.
sudo ./build_sdcard.sh -d headless -f 0
# updating locale ...
*****************************************
*     RASPIBLITZ SD CARD IMAGE SETUP    *
*****************************************
For details on optional parameters - call with '--help' or check source code.
interaction=true
fatpack=false
github_user=raspiblitz
branch=v1.11
display=headless
tweak_boot_drive=true
wifi_region=US
cpu=aarch64
architecture=arm64
baseimage=raspios_arm64
raspi_configfile=/boot/firmware/config.txt
raspi_commandfile=/boot/firmware/cmdline.txt
# Do you agree with all parameters above? (yes/no)

rootzoll, 2025-05-17:
Also retested and found `displayClass='headless'` in raspiblitz.conf - closing issue.

closed 2025-05-17 by rootzoll