raspiblitz/raspiblitz #4975: LND won't start, again (v1.11.4)

Godrigos, 2025-03-20:
Same I've experienced  here #4740 , but now on version 1.11.4.

[Logs](https://termbin.com/oohr).

Thanks for the attention.

Godrigos, 2025-03-30:
Since log is out. And I don't know if anybody get it, here it is:

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.11.4
commit-release: fbcd2b83
commit-active: fbcd2b83
chainnetwork: bitcoin / main
 12:47:15 up 1 day, 18:33,  3 users,  load average: 1.14, 0.88, 0.79

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-03-18 18:17:07 GMT; 1 day 18h ago
    Process: 6751 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6774 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 6891 (bitcoind)
      Tasks: 19 (limit: 9437)
        CPU: 43min 40.352s
     CGroup: /system.slice/bitcoind.service
             └─6891 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Notice: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2025-03-20T12:17:48Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:18:08Z Saw new header hash=00000000000000000001fa13198e019f35251713db164cda4b176c61909dc62f height=888626
2025-03-20T12:18:08Z UpdateTip: new best=00000000000000000001fa13198e019f35251713db164cda4b176c61909dc62f height=888626 version=0x32f00000 log2_work=95.506014 tx=1168420159 date='2025-03-20T12:17:45Z' progress=1.000000 cache=114.5MiB(978752txo)
2025-03-20T12:19:33Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:19:44Z Socks5() connect to ***.onion:8333 failed: host unreachable
2025-03-20T12:19:48Z New block-relay-only v1 peer connected: version: 70016, blocks=888626, peer=2290
2025-03-20T12:20:28Z Saw new header hash=0000000000000000000026ea1e25cf9ae05cafde42164ddc96a72d260f9715ae height=888627
2025-03-20T12:20:28Z UpdateTip: new best=0000000000000000000026ea1e25cf9ae05cafde42164ddc96a72d260f9715ae height=888627 version=0x216c2000 log2_work=95.506026 tx=1168420756 date='2025-03-20T12:20:05Z' progress=1.000000 cache=114.8MiB(979188txo)
2025-03-20T12:20:38Z [i2p] Error connecting to ***.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=pvqyvn2lpvoeyhgcgunoqtetkrkp76iegyoii2af4crnlto6gb2q.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=pvqyvn2lpvoeyhgcgunoqtetkrkp76iegyoii2af4crnlto6gb2q.b32.i2p"
2025-03-20T12:20:55Z [i2p] Error connecting to ***.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=stmwbkh64su2brcqo7pty3mv325mtq55pxibnqhi3ahg7dxeeloq.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=stmwbkh64su2brcqo7pty3mv325mtq55pxibnqhi3ahg7dxeeloq.b32.i2p"
2025-03-20T12:21:44Z [i2p] Error connecting to ***.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=q52c3eye3gvaepqhu45nt4ihgqfuzsr5hh5m5ngqtrhqmwjejjnq.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=q52c3eye3gvaepqhu45nt4ihgqfuzsr5hh5m5ngqtrhqmwjejjnq.b32.i2p"
2025-03-20T12:23:15Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:23:29Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:24:34Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:25:04Z Saw new header hash=0000000000000000000013ab42f555783fcfa68a8b3009541b7201c3e6c8e11b height=888628
2025-03-20T12:25:04Z [net] Saw new cmpctblock header hash=0000000000000000000013ab42f555783fcfa68a8b3009541b7201c3e6c8e11b peer=1325
2025-03-20T12:25:04Z UpdateTip: new best=0000000000000000000013ab42f555783fcfa68a8b3009541b7201c3e6c8e11b height=888628 version=0x200d4000 log2_work=95.506038 tx=1168422066 date='2025-03-20T12:24:51Z' progress=1.000000 cache=114.8MiB(981356txo)
2025-03-20T12:25:40Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:26:45Z [i2p] Error connecting to ***.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER MESSAGE="LeaseSet not found""
2025-03-20T12:27:24Z [i2p] Error connecting to ***.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=eg4j6ftn4bkovxxlazghurrgntw2pmzwvegfnbl3cocdausdawia.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=eg4j6ftn4bkovxxlazghurrgntw2pmzwvegfnbl3cocdausdawia.b32.i2p"
2025-03-20T12:27:41Z Saw new header hash=000000000000000000005a20b4ca34d5c32d4bdbaec1d1bb0ceb4121872b21aa height=888629
[... cut ...]
btc_default_sync_progress="0.999997344004778"
btc_default_blocks_behind="0"
btc_default_sync_initial_done="1"
btc_default_sync_percentage="100.00"
*** POSSIBLE ERROR REPORTS ***

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Smiril, 2025-06-27:
i´ve got similar logs 

same issue @rootzoll