raspiblitz/raspiblitz #4977: Help improve BitBanana connect instructions

michaelWuensch, 2025-03-28:
Hi, I am in the process of writing a nice documentation for BitBanana.
Like Zeus, BitBanana allows you to connect to and manage a node remotely, making RaspiBlitz available on the go.

Unfortunately I have no Raspiblitz so I cannot test it but it shouldn't be to complicated. BitBanana uses gRPC for connection, not REST.

It would be cool if someone that runs a raspiblitz can test it and provide the steps in a list similar to what I have for Umbrel for example.

https://bitbanana.gitbook.io/bitbanana-docs/setup/connect-a-node/umbrel

Furthermore I'd like to know it a Tailscale setup works and which steps are necessary to set it up.

Your help is much appreciated!

NilByte, 2025-03-28:
I have been using Bitbanana on Android for maybe 4 years and it should be straightforward to connect by using the the build in menu options in Raspiblitz.

[image]

[image]

No issues with Tailscale and it makes Bitbanana much more responsive.

Maybe a user with no current installation can make a more detailed how to set up.
 
Thanks for all the great stuff that you have added over the years.

michaelWuensch, 2025-03-31:
@NilByte Thanks, a more detailed step by step guide would be great, but this is already helpful.

@rootzoll ZAP_ANDROID is basically dead, I was the only developer and rebranded it to BitBanana 2 years ago. Zap did not receive any updates since then and will not receive any in the future. I think it can be safely removed or better just be replaced by BitBanana.