raspiblitz/raspiblitz #4983: setting up RPi5/NVMe 2TB, my experiences

JessieRaspi, 2025-04-21:
I recently assembled a new system and would like to share my experiences, since I encountered a few weird things, which I didn’t experienced before with the RPi4 installation with the 1 TB SSD and same display. It might be good to mention those in the user documentation in case other users want to experiment with the same configuration.

My new system components:

RPi 5 8Mb with official power supply
RPi 5 active cooler
Pimeroni NVMe Base
Crucial P3 Plus PCIe 4.0 M.2 2280 SSD
Elegoo 3.5” display
SDcard Samsung ProEndurance 32GB

I flashed version 1.11.4, the minimal version (not the Web UI one). I actually don’t know if I have the right version...it says: 1.11.4-fbcd2b83, same as on my RPi 4.

After assembling the RPi5 hardware parts, inserting the SD card, connecting the internetcable and finally the power, the lights were on, but nothing appeared on the LCD screen (white). I stopped everything, checked the hardware, especially the very tiny ribbon PCIe cable between the RPi5 board and the NVMe base, re-assembled everything and started the booting process again, but the same situation. Screen stayed white. I then flashed another new SD card, but that didn’t help either. 

Then I got the idea to log into my router, to see if the system was recognized in my network. It had an IP address assigned, so I managed to SSH in with the standard pw “raspiblitz” and could proceed from there.

I then followed the dialog screens on the terminal at my PC  and choose to setup a brand new LND Node. 

At some point a strange message appeared:

BOOTSTRAP EXIT ... waiting for final setup controller to initiate final reboot.
------------> You may login thru web browser to continue setup.

I don’t use the webbrowser and waited a very long time, nothing happened and I searched on Github what to do. I found this same message in issue #4541, however it seemed solved. Btw, LCD screen continued to stay white.

I ignored the weird message and after opening a new terminal, SSH’d into the new Raspiblitz with my newly created pw. Finally the steps for the final setup appeared, which were all very clear. 
I chose to copy the blockchain from my Pi4 over LAN (works great btw!) and finally I got the screen working with the command: LCD, so I guess it was disabled the whole time during the bootprocess.
 
Here are my points for the user documentation.

1. Under “Boot your Raspiblitz”: If you have a LCD screen, but it stays white during setup, follow the same procedure as “how do I find the IP address when running without a display” and proceed in the terminal
2. If your screen keeps staying white after setup, then type “LCD” in the command line
3. Ignore the Bootstrap Exit message. Open a new terminal and SSH in with your newly created pw. Then follow the final setup steps.

JessieRaspi, 2025-05-15:
Yes, I am using the NVME as "data" drive and the SD card for booting. 

I originally wanted to combine my system with a simple UPS, since we recently had some power outages, however I struggled to find the right combination with the Pimeroni NVME base and the display on top. 

So, I decided to wait no longer and keep it as simple as possible for the moment!

rootzoll, 2025-06-02:
@JessieRaspi 

> I flashed version 1.11.4, the minimal version [..] but nothing appeared on the LCD screen (white)

This is expected behavior of the mininal SD card image on final releases - minimal also means it does not ship with LCD drivers by default. After setup you can install/activate drivers in the SETTINGS menu or by `lcd` command on CLI. Until then if the LCD hardware is connected - it will just show a white screen.

- [ ] add better info on this in documentation

> At some point a strange message appeared:
> BOOTSTRAP EXIT ... waiting for final setup controller to initiate final reboot.
> ------------> You may login thru web browser to continue setup.

Where/How did the message appeared? Was it in a dialog box, or just plain text ... where you thrown to CLI or was it more of a wait loop?

JessieRaspi, 2025-06-20:
@rootzoll 

The bootstrap exit message appeared as plain text on my pc screen in the terminal, not in a dialog box. I think it was a kind of "wait loop". Since nothing happened, I opened a new terminal and SSH'd in, but this time with my newly created password.