raspiblitz/raspiblitz #4985: Syncing issue

mrpaper87, 2025-05-04:
Sync is blocked and does not go on:

[image]

How i solve it?

ferdinandhodl, 2025-05-05:
checklist (cli commands in paranthesis)
- [ ] process stuck (ps aux)
- [ ] device running full (df -h)
- [ ] check bitcoin chain info (bitcoin-cli getblockchaininfo)
- [ ] rescan bitcoin blocks (bitcoin-cli -rescan)
- [ ] check LN peer list (lncli listpeers)
- [ ] try relaunch syncronisation - reboot?

mrpaper87, 2025-05-06:
ps aux

USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           1  0.2  0.1 169320  9868 ?        Ss   17:11   0:02 /sbin/init
root           2  0.0  0.0      0     0 ?        S    17:11   0:00 [kthreadd]
root           3  0.0  0.0      0     0 ?        S    17:11   0:00 [pool_workqueue_release]
root           4  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-rcu_g]
root           5  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-rcu_p]
root           6  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-slub_]
root           7  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-netns]
root           8  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/0:0-events]
root          11  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/u8:0-ext4-rsv-conversion]
root          12  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-mm_pe]
root          13  0.0  0.0      0     0 ?        I    17:11   0:00 [rcu_tasks_kthread]
root          14  0.0  0.0      0     0 ?        I    17:11   0:00 [rcu_tasks_rude_kthread]
root          15  0.0  0.0      0     0 ?        I    17:11   0:00 [rcu_tasks_trace_kthread]
root          16  0.0  0.0      0     0 ?        S    17:11   0:00 [ksoftirqd/0]
root          17  0.0  0.0      0     0 ?        I    17:11   0:00 [rcu_preempt]
root          18  0.0  0.0      0     0 ?        S    17:11   0:00 [migration/0]
root          19  0.0  0.0      0     0 ?        S    17:11   0:00 [cpuhp/0]
root          20  0.0  0.0      0     0 ?        S    17:11   0:00 [cpuhp/1]
root          21  0.0  0.0      0     0 ?        S    17:11   0:00 [migration/1]
root          22  0.0  0.0      0     0 ?        S    17:11   0:00 [ksoftirqd/1]
root          25  0.0  0.0      0     0 ?        S    17:11   0:00 [cpuhp/2]
root          26  0.0  0.0      0     0 ?        S    17:11   0:00 [migration/2]
root          27  0.0  0.0      0     0 ?        S    17:11   0:00 [ksoftirqd/2]
root          28  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/2:0-pm]
root          30  0.0  0.0      0     0 ?        S    17:11   0:00 [cpuhp/3]
root          31  0.0  0.0      0     0 ?        S    17:11   0:00 [migration/3]
root          32  0.0  0.0      0     0 ?        S    17:11   0:00 [ksoftirqd/3]
root          35  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/u9:0-events_unbound]
root          36  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/u10:0-events_unbound]
root          38  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/u12:0-events_unbound]
root          39  0.0  0.0      0     0 ?        S    17:11   0:00 [kdevtmpfs]
root          40  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-inet_]
root          42  0.0  0.0      0     0 ?        S    17:11   0:00 [kauditd]
root          43  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/0:2-pm]
root          44  0.0  0.0      0     0 ?        S    17:11   0:00 [khungtaskd]
root          45  0.0  0.0      0     0 ?        S    17:11   0:00 [oom_reaper]
root          46  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-write]
root          47  0.0  0.0      0     0 ?        S    17:11   0:00 [kcompactd0]
root          48  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-kinte]
root          49  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-kbloc]
root          50  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-blkcg]
root          52  0.0  0.0      0     0 ?        S    17:11   0:00 [watchdogd]
root          54  0.0  0.0      0     0 ?        I    17:11   0:00 [kworker/3:1-mm_percpu_wq]
root          55  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/3:1H-kblockd]
root          56  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-rpcio]
root          57  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-xprti]
root          59  0.4  0.0      0     0 ?        S    17:11   0:04 [kswapd0]
root          60  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-nfsio]
root          61  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-kthro]
root          64  0.0  0.0      0     0 ?        S    17:11   0:00 [hwrng]
root          65  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-iscsi]
root          66  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-nvme-]
root          67  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-nvme-]
root          68  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-nvme-]
root          69  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-DWC N]
root          70  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-uas]
root          71  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/u13:0-brcmf_wq/mmc1:0001:1]
root          72  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/u14:0]
root          73  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/u15:0]
root          74  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/u16:0]
root          75  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/u17:0]
root          76  0.0  0.0      0     0 ?        S    17:11   0:00 [irq/36-aerdrv]
root          77  0.0  0.0      0     0 ?        S    17:11   0:00 [irq/42-aerdrv]
root          79  0.0  0.0      0     0 ?        I<   17:11   0:00 [kworker/R-sdhci]
[... cut ...]
rescan bitcoin blocks (bitcoin-cli -rescan)

not working

--------------------------------------------------------------------------------------------------------------------------------------------

lncli listpeers

{
    "peers": []
}

--------------------------------------------------------------------------------------------------------------------------------------------

try relaunch syncronisation - reboot?

Yes but not working, I have already synchronised twice from the beginning but it does not complete.

closed 2025-05-11 by mrpaper87

ferdinandhodl, 2025-05-12:
report how you solved it

mrpaper87, 2025-05-12:
the third time I reset and synchronised again the blockchain completed the process