raspiblitz/raspiblitz #4986: Raspi doesn't like Bitaxe 601 Miner?

NilsGermany, 2025-05-08:
Hi All,

in short: 
My Bitaxe 601, connected in my home LAN to my Raspiblitz (Raspi 5, 8 MB, 2TB SSD), switches to fallback STRATUM address after approx 15-20 minutes every time. 
I did a 3 day test against eusolo.ckpool.org (reachable via internet and not even in my local network) and there was no fallback case. :-o

In long:
- no WLAN or network issues in FRITZ.BOX router detected
- BitAxe (V 27.1) does not overheat
- not much traffic in my local network
-  RaspiBlitz 1.11.4-fbcd2b83  Goldleader
               Bitcoin Fullnode + Lightning Network + Tor
        ,/     -------------------------------------------
      ,'/      Refreshed: 08 May 11:28:39 +0200 up 4d
    ,' /       CPU load 1.38, 1.37, 1.56 temp 53°C 127°F
  ,'  /_____   Free Mem 5500M / 7945M  HDD 2TB 44%
,'_____    ,'  SSH admin@XXXXXX d6.5GiB u10.1GiB
      /  ,'    Web Admin --> http://XXXXX
     / ,'      bitcoin v27.2.0 mainnet 16 peers
    /,'        Blocks 895790/895790 Sync OK 100.00%

Any idea what the issue might be?

Thanks and regards from Germany
Nils

NilByte, 2025-05-08:
If you run debug in terminal, what does the `*** LAST 20 PUBLIPOOL LOGS ***` show?

There is a suggestion to increase the rpcworkqueue to 512 in bitcoin.conf in this thread:

https://github.com/raspiblitz/raspiblitz/issues/4903

I made this change for rpcworkqueue  and also updated the npm version to 11.2.0.

Run npm version in terminal to get current npm version to get back to it if the change cause problems.

Then update npm version:

`sudo npm install -g npm@11.2.0`

Restart pool:
`sudo systemctl restart publicpool`

Restart ui:
`sudo systemctl restart publicpool-ui`

Get status pool:
`systemctl status publicpool`

Get status ui:
`systemctl status publicpool-ui`

I'm currently reaching max one week uptime without pool restarting but besides this it seems to be working quite stable running on Pi5, 8GB.

NilsGermany, 2025-05-08:
Hello NilByte 
thanks for the quick reply!

Where do I find any publicpool log files on my Raspi?  

I do not have a line with parameter rpcworkqueue in bitcoin.conf. Should I just insert it and if so, in which section?

admin@192.168.178.70:~ ₿ npm version
{
  npm: '10.8.2',
  node: '20.18.3',
  acorn: '8.14.0',
  ada: '2.9.2',
  ares: '1.34.4',
  base64: '0.5.2',
  brotli: '1.1.0',
  cjs_module_lexer: '1.4.1',
  cldr: '45.0',
  icu: '75.1',
  llhttp: '8.1.2',
  modules: '115',
  napi: '9',
  nghttp2: '1.61.0',
  nghttp3: '0.7.0',
  ngtcp2: '1.1.0',
  openssl: '3.0.15+quic',
  simdutf: '5.6.4',
  tz: '2024b',
  undici: '6.21.1',
  unicode: '15.1',
  uv: '1.46.0',
  uvwasi: '0.0.21',
  v8: '11.3.244.8-node.25',
  zlib: '1.3.0.1-motley-82a5fec'
}

admin@192.168.178.70:~ ₿ sudo systemctl status publicpool
● publicpool.service - publicpool
     Loaded: loaded (/etc/systemd/system/publicpool.service; enabled; preset: enabled)
     Active: active (running) since Thu 2025-05-08 14:50:55 CEST; 4h 31min ago
   Main PID: 520482 (npm start)
      Tasks: 35 (limit: 9438)
        CPU: 1min 56.314s
     CGroup: /system.slice/publicpool.service
             ├─520482 "npm start"
             ├─520493 sh -c "nest start"
             ├─520494 node /home/publicpool/publicpool/node_modules/.bin/nest start
             ├─520710 /bin/sh -c "node -r source-map-support/register /home/publicpool/publicpool/dist/main"
             └─520711 node -r source-map-support/register /home/publicpool/publicpool/dist/main

May 08 18:28:49 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:49 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:50 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:50 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:51 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:51 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:52 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:52 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:53 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"
May 08 18:28:53 raspberrypi npm[520711]: Error getmininginfo 503 - "Work queue depth exceeded"

NilByte, 2025-05-08:
In terminal run the command `debug` and look for the  *** LAST 20 PUBLIPOOL LOGS *** section.

If you don't have these two lines in bitcoin.conf, you can add them as last lines:

```
rpcworkqueue=512
rpcthreads=128
```

I think you can ignore the WARN: Missing $DBUS_SESSION_BUS_ADDRESS .. in your troubleshooting.

NilsGermany, 2025-05-08:
Here we go...
I will now insert the stated parameters to bitcoin.conf and cross my fingers... ;-)

*** LAST 20 PUBLIPOOL LOGS ***
sudo journalctl -u publicpool -b --no-pager -n20
May 08 21:03:28 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:28 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:29 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:29 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:30 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:30 raspberrypi npm[2128836]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 08 21:03:31 raspberrypi npm[2128836]: Error getblocktemplate: Error: ESOCKETTIMEDOUT
May 08 21:03:31 raspberrypi npm[2128836]: /home/publicpool/publicpool/src/services/bitcoin-rpc.service.ts:131
May 08 21:03:31 raspberrypi npm[2128836]:             throw new Error('Error getblocktemplate');
May 08 21:03:31 raspberrypi npm[2128836]:                   ^
May 08 21:03:31 raspberrypi npm[2128836]: Error: Error getblocktemplate
May 08 21:03:31 raspberrypi npm[2128836]:     at BitcoinRpcService.getBlockTemplate (/home/publicpool/publicpool/src/services/bitcoin-rpc.service.ts:131:19)
May 08 21:03:31 raspberrypi npm[2128836]:     at processTicksAndRejections (node:internal/process/task_queues:95:5)
May 08 21:03:31 raspberrypi systemd[1]: publicpool.service: Main process exited, code=exited, status=1/FAILURE
May 08 21:03:31 raspberrypi systemd[1]: publicpool.service: Failed with result 'exit-code'.
May 08 21:03:31 raspberrypi systemd[1]: publicpool.service: Consumed 27.861s CPU time.
May 08 21:04:01 raspberrypi systemd[1]: publicpool.service: Scheduled restart job, restart counter is at 3.
May 08 21:04:01 raspberrypi systemd[1]: Stopped publicpool.service - publicpool.
May 08 21:04:01 raspberrypi systemd[1]: publicpool.service: Consumed 27.861s CPU time.
May 08 21:04:01 raspberrypi systemd[1]: Started publicpool.service - publicpool.

NilsGermany, 2025-05-09:
Update from now (with inserted rpc parameters):

*** LAST 20 PUBLIPOOL LOGS ***
sudo journalctl -u publicpool -b --no-pager -n20
May 09 08:52:07 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:08 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:08 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:09 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:09 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:10 raspberrypi npm[445407]: Error getmininginfo Error: ESOCKETTIMEDOUT
May 09 08:52:10 raspberrypi npm[445407]: Error getblocktemplate: Error: ESOCKETTIMEDOUT
May 09 08:52:10 raspberrypi npm[445407]: /home/publicpool/publicpool/src/services/bitcoin-rpc.service.ts:131
May 09 08:52:10 raspberrypi npm[445407]:             throw new Error('Error getblocktemplate');
May 09 08:52:10 raspberrypi npm[445407]:                   ^
May 09 08:52:10 raspberrypi npm[445407]: Error: Error getblocktemplate
May 09 08:52:10 raspberrypi npm[445407]:     at BitcoinRpcService.getBlockTemplate (/home/publicpool/publicpool/src/services/bitcoin-rpc.service.ts:131:19)
May 09 08:52:10 raspberrypi npm[445407]:     at processTicksAndRejections (node:internal/process/task_queues:95:5)
May 09 08:52:10 raspberrypi systemd[1]: publicpool.service: Main process exited, code=exited, status=1/FAILURE
May 09 08:52:10 raspberrypi systemd[1]: publicpool.service: Failed with result 'exit-code'.
May 09 08:52:10 raspberrypi systemd[1]: publicpool.service: Consumed 1min 58.465s CPU time.
May 09 08:52:40 raspberrypi systemd[1]: publicpool.service: Scheduled restart job, restart counter is at 12.
May 09 08:52:40 raspberrypi systemd[1]: Stopped publicpool.service - publicpool.
May 09 08:52:40 raspberrypi systemd[1]: publicpool.service: Consumed 1min 58.465s CPU time.
May 09 08:52:40 raspberrypi systemd[1]: Started publicpool.service - publicpool.

Any other hint how I could fix it?
As said, I'm running this setup in my LAN with no latency and not much other traffic.

NilsGermany, 2025-05-09:
Sorry, me again. I found this as a solution, but I am not sure where to add this parameter?  

_My public pool instance just trows an error and crash in a RPC call timeout. **You can try increasing BITCOIN_RPC_TIMEOUT in .env file to avoid timeouts. I just changed it to 1000 seconds, since i run all locally.**_

NilsGermany, 2025-05-10:
Hello NilByte!

I guess your last post was the final clue! Now I'm at more than 1k shares, seems stable.
Thanx again.

In a nutshell what I did:
- updated the npm version to 11.2.0.
- Changes to Bitcoin.conf:
        rpcworkqueue=512
        rpcthreads=128
        rpctimeout=10000
        timeout=10000
- /home/publicpool/publicpool/.env --> BITCOIN_RPC_TIMEOUT=100000 (one hundert thousend!)

All the best from Germany
Nils

NilByte, 2025-06-13:
Seems like the trick to get rid of ESOCKETTIMEDOUT is to increase the  BITCOIN_RPC_TIMEOUT in /home/publicpool/publicpool/.env.

I changed the default value of BITCOIN_RPC_TIMEOUT from 10000 to 60000 and have not had any notices in debug log since then. Uninterrupted uptime now 1 month.

Inspiration for making the BITCOIN_RPC_TIMEOUT configurable from Public Pool for StartOS:

https://github.com/remcoros/public-pool-startos/commit/c3cbe9f6d856bc7c532251d01a06670afc3e90a3

rootzoll, 2026-02-01:
OK so it looks this will be solved with upcoming v1.12.1 .. closing for release.