raspiblitz/raspiblitz #4999: WARN: Missing $DBUS_SESSION_BUS_ADDRESS

NilsGermany, 2025-05-14:
Hello all,

at every reboot or shutdown I get this warning:
WARN: Missing $DBUS_SESSION_BUS_ADDRESS ..
RUNNING FALLBACK REBOOT ..

What do I have to do to get rid of this?
Reboot takes long and I have to start some services via shell / bash.

Can this parameter set to a default value by the system or how can I prevent a FALLBACK reboot?

Thanx and regards

Nils

rootzoll, 2025-06-02:
Hi. Because of time contrains will take a look again after v1.12.0 release

NilsGermany, 2025-06-15:
Hello!
While I know this is open source, again the Raspi doesn't boot up correctly after a normal reboot command:

RaspiBlitz 1.11.4-fbcd2b83  XXX
               Bitcoin Fullnode + Lightning Network + Tor
        ,/     -------------------------------------------
      ,'/      Refreshed: 15 Jun 21:48:41 +0200 up 27m
    ,' /       CPU load 1.65, 2.01, 1.84 temp 50°C 122°F
  ,'  /_____   Free Mem 7387M / 7945M  HDD 2TB 45%
,'_____    ,'  SSH admin@192.XXX d5.1MiB u7.8MiB
      /  ,'    Web Admin --> http://192.XXX
     / ,'      bitcoin v27.2.0 mainnet
    /,'        Not Started | Not Ready Yet | No Data
   /'
               LND 0.18.3-beta
               Not Started | Not Ready Yet | No Data

Returning to menu .....
admin@192.168.178.70:~ ? sudo systemctl status lnd.service
? lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; preset: enabled)
     Active: inactive (dead)
admin@192.168.178.70:~ ? sudo systemctl status bitcoind.service
? bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: inactive (dead)

Jun 15 21:39:18 raspberrypi admin[68669]: FAIL: network.txindex.sh status needs sudo
Jun 15 21:39:25 raspberrypi sudo[68904]:    admin : TTY=pts/1 ; PWD=/home/admin ; USER=root ; COMMAND=/usr/bin/ls /mnt/>
Jun 15 21:39:25 raspberrypi sudo[68904]: pam_unix(sudo:session): session opened for user root(uid=0) by admin(uid=1001)
Jun 15 21:39:25 raspberrypi sudo[68904]: pam_unix(sudo:session): session closed for user root
Jun 15 21:39:25 raspberrypi sudo[68909]:    admin : TTY=pts/1 ; PWD=/home/admin ; USER=root ; COMMAND=/usr/bin/ls /mnt/>
Jun 15 21:39:25 raspberrypi sudo[68909]: pam_unix(sudo:session): session opened for user root(uid=0) by admin(uid=1001)
Jun 15 21:39:25 raspberrypi sudo[68909]: pam_unix(sudo:session): session closed for user root
Jun 15 21:39:25 raspberrypi admin[68916]: FAIL: network.txindex.sh status needs sudo

I used this commands (as in the past) but no result:

LND not starting:
sudo systemctl restart lnd
sudo systemctl status lnd.service

sudo systemctl start bitcoind.service
sudo systemctl status bitcoind.service

Failed:
journalctl -xe
systemctl reset-failed

Unlock Wallet:
lncli unlock

Any hint? Hardware is fine...

BTW: Could anybody who reads this send 10 Sats to this address for a test? Thanx! amber660@ghst.to