raspiblitz/raspiblitz #5003: Open BTC Address in Mempool results in "500 Internal Server Error"

joel-st, 2025-05-16:
**Describe the bug**
The Mempool extension on RaspiBlitz throws a "500 Internal Server Error: Connection to server lost, please retry" when attempting to open any specific Bitcoin address page. While the main Mempool page loads correctly, navigating to a detailed view of an address consistently results in this error. The Nginx access logs show a "500" status code for the API request to fetch address data. But the Nginx Mempool error logs shows nothing.

```
Error loading address data.
(500 Internal Server Error: Connection to server lost, please retry)
```

**To Reproduce**
Run Raspiblitz, install Mempool, try to open any specific Bitcoin address page.

**Debug file**
https://termbin.com/mj3p

Thanks for all the effort maintaining this repo! Raspiblitz is super nice 🙏🏻

rootzoll, 2025-06-02:
Hi. On which RaspiBlitz/Mempool Version are you gettung this error?

joel-st, 2025-06-03:
Hey, thanks for getting in touch!

Versions
RaspiBlitz: 1.11.4
RaspiBlitz WebUI: 1.3.2
Blitz-API: v0.5.1beta
Mempool: v3.0.0

rootzoll, 2025-06-03:
Was also able to replicate this now on RaspuBlitz v1.12.0rc3 .. will take a deeper look.

rootzoll, 2025-06-03:
Ah wait - electrs is not installed, right? mempool needs electrs index to display single addresses.

closed 2025-06-04 by joel-st