raspiblitz/raspiblitz #5007: misleading startup message

ferdinandhodl, 2025-05-17:
**Describe the bug**
After starting up Raspiblitz, the LCD shows `ssh admin@192.168.0.42 -> Password A` which is wrong and misleading. Accessing Raspiblitz via _webinterface_ (e.g. http://192.168.0.42 or https://192.168.0.42 respectively) would indeed ask for **Password A** while the _CLI_ asks for **Password C** in order to unlock the wallet. This minor bug is misleading and discourages new users, eventually causing them to question reliability of Raspiblitz. Further this behavior represents a security issue, as a password working for another service is entered/submitted to the wrong service.

**To Reproduce**
Restart Raspiblitz, follow display instructions

**Version**
- RaspiBlitz: 1.11.4
- RaspiBlitz WebUI: 1.3.2
- Blitz-API: v0.5.1beta

**tags**
- UX / User eXperience
- security

ferdinandhodl, 2025-06-03:
please find the attached screenshot CLI vs. LCD dialog

[image]

rootzoll, 2025-06-04:
Yeah OK I see what you mean .. one solution could be that it should be to display on the LCD instead:

```
Login to unlock Lightning Wallet
ssh admin@10.42.0.107
Use Password A for login
Use Password C to unlock wallet
```