raspiblitz/raspiblitz #5026: [1.12.0] Install JAM fails

rootzoll, 2025-05-20:
@openoms tested install of JAM on v1.12.0rc2 and it fails. To be tested with rc3

```
src/libsodium/Makefile.am: installing 'build-aux/depcomp'
autoreconf: Leaving directory '.'
Downloading config.guess and config.sub...
Done.
checking build system type... ./build-aux/config.guess: line 1: html: No such file or directory
./build-aux/config.guess: line 2: syntax error near unexpected token `<'
'/build-aux/config.guess: line 2: `<head><title>502 Bad Gateway</title></head>
configure: error: cannot guess build type; you must specify one
gmake: *** No rule to make target 'uninstall'.  Stop.
gmake: *** No targets specified and no makefile found.  Stop.
gmake: *** No rule to make target 'check'.  Stop.
Libsodium was not built. Exiting.
 Failed to install JoinMarket
```

fusion44, 2025-05-23:
I think I can confirm. At least something is wrong with the function on the script. I've installed JAM via the webui without any indication that it failed and the app doesn't appear as installed. I then went to check the status via SSH and I get the following:
```
> ./config.scripts/bonus.jam.sh status
version='0.3.0'
installed='0'
localIP='192.168.8.197'
httpPort='7500'
httpsPort='7501'
httpsForced='1'
httpsSelfsigned='1'
authMethod='password_b'
toraddress=''
```
Installed is still 0.

fusion44, 2025-05-23:
When I run the installer manually:
```
admin@192.168.8.197:~ ₿ ./config.scripts/bonus.jam.sh on
# isActive(0)
# isInstalled(1)
# OK, running behind Tor
# setting PASSWORD_B as the password for the 'joinmarket' user
# Ok, Joininbox is present

#############
# Autostart #
#############

if [ -f "/home/joinmarket/joinmarket-clientserver/jmvenv/bin/activate" ]; then
  . /home/joinmarket/joinmarket-clientserver/jmvenv/bin/activate
  cd /home/joinmarket/joinmarket-clientserver/scripts/
fi
# shortcut commands
source /home/joinmarket/_commands.sh
# automatically start main menu for joinmarket unless
# when running in a tmux session
if [ -z "$TMUX" ]; then
  /home/joinmarket/menu.sh
fi

# Check 'deprecatedrpc=create_bdb' in bitcoin.conf
deprecatedrpc=create_bdb
# The bitcoin core wallet is already on
# Restarting bitcoind
# The non-descriptor wallet.dat is loaded in bitcoind.
# starting JoininBox ...
/home/joinmarket/_functions.sh: line 21: /home/joinmarket/standalone/_functions.standalone.sh: No such file or directory
sudo: /home/joinmarket/standalone/expand.rootfs.sh: command not found
# Generating joinmarket.cfg with default settings

Traceback (most recent call last):
  File "/home/joinmarket/joinmarket-clientserver/scripts/wallet-tool.py", line 2, in <module>
    from jmbase import jmprint
ModuleNotFoundError: No module named 'jmbase'
chmod: cannot access '/home/joinmarket/.joinmarket/joinmarket.cfg': No such file or directory
# There was an error running 'bonus.joinmarket.sh on', see above
```

openoms, 2025-06-06:
Will also need to adapt to the new location of the bitcoin.conf.

PR #5048 (2025-06-07): update joininbox to v0.8.4
fixes: #5026
closed 2025-06-07 by rootzoll