raspiblitz/raspiblitz #5037: [1.12.0rc3] Electrs fail

geco91, 2025-05-31:
Webinterface ist ElectRs rot und sagt unable to execute the bash script /home/admin/config.scripts/bonus.electrs.sh status ├╴at /home/blitzapi/blitz_api/app/api/utils.py:533:10 ├╴command: /home/admin/config.scripts/bonus.electrs.sh status │ ╰─▶️ ├╴at /home/blitzapi/blitz_api/app/api/utils.py:533:10 │ ╰╴backtrace (2). Das genannte script selbst läuft aber manuell aufgerufen sauber durch.

Schnema1, 2025-06-06:
Not sure what is the exact problem. 

While on an other PR (rust installed system wide for cln), I could not reproduce your issue. If I remember correct, bitcoin must be synced 100% in order to get electrs build its db correctly. 

In my case I had bitcoin synced 100% with core lightning and Web UI installed. To try reproduce your error, I chose the the ElectRS installer from within the Web UI. However it worked and electrs shows as installed. 

Hint: I did a propre shutdown during the sync because I moved the raspberry pi 5 with passive cooler, close to a fan to cool it down. Have you tried a reboot?

Later I can try to reflash the RC3 in order to get a clean state. It did not work, and the elect rs installation was already in progress.

rootzoll, 2025-07-30:
@fusion44 is this something to be added to the blitzAPI before final v1.12.0 release?

> Maybe the error occurs in the file /home/blitzapi/blitz_api/app/api/utils.py: line 533
> I did increase the timeout in line 470 to 20.0 and did a reboot.

fusion44, 2025-08-08:
I've updated the default timeouts from 10 secs to 15 secs. I noticed on my old PI4 when it's under full load that it'll also stop some queries to early occasionally. I've increased the timeout for Electrs to 30 secs. We must be aware though, that these queries do not run in parallel. On high load systems it could even further increase the response times. 

It's all pushed to the 1.12 branch.

closed 2025-08-09 by rootzoll