raspiblitz/raspiblitz #5128: tail: cannot open ...

rootzoll, 2025-09-22:
Reported on support chat

[image]

not critical but cosmetic to fix - check to mute error if tail fails

elsteff1385, 2025-10-07:
Same error in WebUI in BTC RPC Explorer app

[image]

elsteff1385, 2025-10-07:
Also for Mempool.space app

[image]

rootzoll, 2025-10-22:
OK I merged #5156 into v1.12 branch .. please run `SSHMENU > UPDATE > PATCH` & restart and let me know if that fixes this issue.